ToFuProject/tofu #62: Nosetest fail since IMAS module was added

lasofivec, 2019-05-07:
Since I merged with the latests version of `master` (with the new module for imas) I get the following error when doing `python setup.py nosetests`. I don't get it when just calling `nosetests [options]`. It seems when nosetests is ran from `distutils` there can be this kind of problems (see this [SO post](https://stackoverflow.com/questions/34400481/setup-py-nosetests-command-throws-importerror-when-specifying-package-dir)).
Error:

> ======================================================================
> ERROR: Failure: Exception (IMAS python API issue
> imas could not be imported into tofu ('import imas' failed):
>   - it may not be installed (optional dependency)
>   - or you not have loaded the good working environment
> 
>     => the optional sub-package tofu.imas2tofu is not usable
> )
> ----------------------------------------------------------------------
> Traceback (most recent call last):
>   File "/home/u2/mendoza/.conda/envs/tofu3/lib/python3.7/site-packages/nose/failure.py", line 39, in runTest
>     raise self.exc_val.with_traceback(self.tb)
>   File "/home/u2/mendoza/.conda/envs/tofu3/lib/python3.7/site-packages/nose/loader.py", line 417, in loadTestsFromName
>     addr.filename, addr.module)
>   File "/home/u2/mendoza/.conda/envs/tofu3/lib/python3.7/site-packages/nose/importer.py", line 47, in importFromPath
>     return self.importFromDir(dir_path, fqname)
>   File "/home/u2/mendoza/.conda/envs/tofu3/lib/python3.7/site-packages/nose/importer.py", line 94, in importFromDir
>     mod = load_module(part_fqname, fh, filename, desc)
>   File "/home/u2/mendoza/.conda/envs/tofu3/lib/python3.7/imp.py", line 244, in load_module
>     return load_package(name, filename)
>   File "/home/u2/mendoza/.conda/envs/tofu3/lib/python3.7/imp.py", line 216, in load_package
>     return _load(spec)
>   File "<frozen importlib._bootstrap>", line 696, in _load
>   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
>   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
>   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
>   File "/home/u2/mendoza/tofu/tofu/imas2tofu/__init__.py", line 16, in <module>
>     raise Exception(msg)
> Exception: IMAS python API issue
> imas could not be imported into tofu ('import imas' failed):
>   - it may not be installed (optional dependency)
>   - or you not have loaded the good working environment
> 
>     => the optional sub-package tofu.imas2tofu is not usable
>

Didou09, 2019-06-17:
@lasofivec can you definitely confirm that your pull request solved this issue on your laptop (I think the issue was only showing on your laptop or Mac) ?
If yes it can be closed :-)

lasofivec, 2019-06-18:
Yes, it was on all my machines, and the issue is now resolved :)

PR #98 (2019-06-17): Issue62 nosetest fail
Made two changes:
- libraries that need imas2tofu are only imported if the imas module is found
- instead of raising an error, we just raise a Warning

Fixes #62
closed 2019-06-18 by lasofivec