ToFuProject/tofu #73: Solid angles particle - NxM points no unit vector - cython for large plasma sub-domain toroidal integral

Didou09, 2019-05-23:
Cython is necessary here for memory / speed performance.

We will intergate toroidally over a potentially large plasma sub-domain (or the whole domain).
It implies:
* Determining the plasma volume to be sampled (from user input)
* Sampling the volume (copy the already existing algorithm in _GG02.pyx)
* As the volume is being sampled, include a few extra steps:
      * sampling has to be done by looping on R (outer main loop), because R determines the number of toroidal points
      * then loop on Z
      * and integrate toroidally by updating a cumulative toroidal sum of solid angles

The output should be a 2D (R,Z) map of toroidally integrated solid angles (one map for each M point on the trajectory, so in fact the output should be a (M,R,Z) array).

The question of where to put the  loop on the M points remains to be determined.
Maybe inside (innermost loop) so the parallelization is done on R ?

lasofivec, 2021-02-23:
With "sub domain" you mean there should be a vignetting done right ?

Didou09, 2021-02-23:
"Sub-domain" refers to the fact that the plasma volume is defined like for the plasma volume sampling routines: from sub-domains (intervals in R, Z, phi) or from indices.
The idea is realy to re-use what you've done already, same philosophy.

And yes, there should be some vignetting done from structural elements.
An example of that is in tofu.geom._comp_solidangles.py