ToFuProject/tofu #84: Bug in _Ves_Vmesh_Tor_SubFromD_cython:  over-indexing in array

lasofivec, 2019-05-27:
There is a problem in the following line that somehow only show on my Mac

https://github.com/ToFuProject/tofu/blob/c5fde53717d9f8a900ddfcf7c1eb9d695d6b34e2/tofu/geom/_GG02.pyx#L806-807

The arrays `R` and `Rn` are of the same size (`Rp`). However `dRPhir` is of size `Rn`.
So if `Rp > Rn`, we are for sure accessing memory that wasn't allocated. 

The error can be detected by adding: 
`@cython.boundscheck(False)`
before the function....

Didou09, 2019-05-28:
OK, thanks for the notice, I'll have a look asap (this afternoon or tomorrow)

lasofivec, 2019-07-05:
======================================================================
ERROR: tofu.tests.tests01_geom.tests03_core.Test01_Struct.test13_get_sampleV
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/mendoza/tofu/tofu/tests/tests01_geom/tests03_core.py", line 372, in test13_get_sampleV
    Out='(X,Y,Z)')
  File "/Users/mendoza/tofu/tofu/geom/_core.py", line 885, in get_sampleV
    pts, dV, ind, reseff = _comp._Ves_get_sampleV(*args, **kwdargs)
  File "/Users/mendoza/tofu/tofu/geom/_comp.py", line 263, in _Ves_get_sampleV
    Pts, dV, ind, dVr[0], dVr[1], dVr[2] = _GG._Ves_Vmesh_Tor_SubFromD_cython(dV[0], dV[1], dV[2], MinMax1, MinMax2, DR=DV[0], DZ=DV[1], DPhi=DV[2], VPoly=VPoly, Out=Out, margin=margin)
  File "tofu/geom/_GG03.pyx", line 1032, in tofu.geom._GG03._Ves_Vmesh_Tor_SubFromD_cython
    dRPhir = np.array([dRPhir[ii] for ii in range(0,len(R)) \
IndexError: Out of bounds on buffer access (axis 0)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/mendoza/anaconda3/envs/tofu3/lib/python3.7/site-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/Users/mendoza/tofu/tofu/tests/tests01_geom/tests03_core.py", line 384, in test13_get_sampleV
    raise Exception(msg)
Exception: Out of bounds on buffer access (axis 0)
Failed for Tor_Ves_VesIn
    ii=0
    Lim=[]
    DS=None

PR #110 (2019-07-05): Issue109
## Fixed three Issues:

- Un registered issue: wrong size of VPoly for a test computing kMin kMax (bug in test part)
- Fixes #84 : after discussing with @Didou09 commented a part that's probably not necessary and fixed error of redundant variable name
- Fixes #109 : I was sending wrong size of
PR #111 (2019-07-05): Fixed three issues
## Fixed three Issues:

- Un registered issue: wrong size of VPoly for a test computing kMin kMax (bug in test part)
- Fixes #84 : after discussing with @Didou09 commented a part that's probably not necessary and fixed error of redundant variable name
- Fixes #109 : I was sending wrong size of
closed 2019-07-05 by lasofivec