ToFuProject/tofu #89: Compilation issue on IRFM clusters

Didou09, 2019-06-11:
There seems to be a compilation issue after the last merge.
From the error messages (see below), it looks like the compiler simply does not like it when a line is broken.

Laura, could you give us your opinion ? Did you encounter anything like that ?
It is strange that this issue did not show during tests,
It may be because the linux clusters at IRFM are using gcc with an old version of GLIBC (on Cent OS 6.9) whereas Travis is running a modern ubuntu with modern GLIBC.

----------------- Error message ---------------------------------------------

(base) [ DV226270 sirrah ~/ToFu_All/tofu_git/tofu (Issue61_Imas2tofuMoreRobust)] python setup.py build_ext --inplace

Version for setup.py :  1.4.0-138-g9c4d8e8

Using Cython !!!!!!!!!

Compiling tofu/geom/_GG03.pyx because it changed.
Compiling tofu/geom/_basic_geom_tools.pyx because it changed.
[1/2] Cythonizing tofu/geom/_GG03.pyx

Error compiling Cython file:
------------------------------------------------------------
...

# ==============================================================================
# =  Computing Hypothenus
# =============================================================================
cdef inline array compute_hypot(double[::1] xpts, double[::1] ypts,
                                int npts=-1):
                               ^
------------------------------------------------------------

tofu/geom/_basic_geom_tools.pxd:136:32: default values cannot be specified in pxd files, use ? or *

Error compiling Cython file:
------------------------------------------------------------
...

# ==============================================================================
# =  Computing Hypothenus
# =============================================================================
cdef inline array compute_hypot(double[::1] xpts, double[::1] ypts,
                                int npts=-1):
                                         ^
------------------------------------------------------------

tofu/geom/_basic_geom_tools.pxd:136:42: Expected ')', found 'INT'

Error compiling Cython file:
------------------------------------------------------------
...
from libc.math cimport atan2 as Catan2, pi as Cpi
from libc.math cimport NAN as Cnan
from libc.math cimport isnan as Cisnan
from libc.stdlib cimport malloc, free
# -- ToFu library imports ------------------------------------------------------
from _basic_geom_tools cimport _VSMALL, _SMALL
^
------------------------------------------------------------

tofu/geom/_GG03.pyx:31:0: 'tofu/geom/_basic_geom_tools/_VSMALL.pxd' not found

Error compiling Cython file:
------------------------------------------------------------
...
from libc.math cimport atan2 as Catan2, pi as Cpi
from libc.math cimport NAN as Cnan
from libc.math cimport isnan as Cisnan
from libc.stdlib cimport malloc, free
# -- ToFu library imports ------------------------------------------------------
from _basic_geom_tools cimport _VSMALL, _SMALL
^
------------------------------------------------------------

tofu/geom/_GG03.pyx:31:0: 'tofu/geom/_basic_geom_tools/_SMALL.pxd' not found

Error compiling Cython file:
------------------------------------------------------------
...
# ==============================================================================
cdef inline bint inter_ray_aabb_box(const int[3] sign,
                                    const double[3] inv_direction,
                                    const double[6] bounds,
                                    const double[3] ds,
                                    bint countin=False,
                                   ^
------------------------------------------------------------

tofu/geom/_raytracing_tools.pxd:29:36: default values cannot be specified in pxd files, use ? or *

Error compiling Cython file:
------------------------------------------------------------
...
cdef inline bint inter_ray_aabb_box(const int[3] sign,
                                    const double[3] inv_direction,
                                    const double[6] bounds,
                                    const double[3] ds,
                                    bint countin=False,
                                    bint debug_plot=False) nogil:
                                   ^
------------------------------------------------------------

tofu/geom/_raytracing_tools.pxd:30:36: default values cannot be specified in pxd files, use ? or *

Didou09, 2019-06-11:
Additional info:

- If I checkout to branch **devel**, the compilation works fine on IRFM clusters (tested today)....

- On travis, the compilation of branch [Issue61_Imas2tofuMoreRobust](https://github.com/ToFuProject/tofu/tree/Issue61_Imas2tofuMoreRobust) works fine (tested [today](https://travis-ci.org/ToFuProject/tofu/builds/544161078))

- The problem only arises when I compile from branch [Issue61_Imas2tofuMoreRobust](https://github.com/ToFuProject/tofu/tree/Issue61_Imas2tofuMoreRobust) on IRFM clusters. Could you try pulling this specific branch on your laptop and trying compiling ?

Didou09, 2019-06-11:
git diff confirms that there is no difference in the tofu/geom directory between [devel](https://github.com/ToFuProject/tofu/tree/devel) and [Issue61_Imas2tofuMoreRobust](https://github.com/ToFuProject/tofu/tree/Issue61_Imas2tofuMoreRobust)

**> git diff Issue61_Imas2tofuMoreRobust devel -- tofu/geom/**
=> returns nothing (i.e.: no changes detected)

Conclusion : the problem is here (on devel) since the merge from DistPoly, but was not detected because the compilation works fine on Travis, but not on IRFM clusters (and I had not tried it, mea culpa)

Didou09, 2019-06-11:
A commit-by-commit series of compilation tests reveals that the problem appeared between:

- commit c5fde53717d9f8a900ddfcf7c1eb9d695d6b34e2 (master)

- 443f112cf36604c2249fbeb1a315e55911fb58f1  (creation of branch DistPolyLOS)

Differences reveal:... many differences in tofu/geom/

Any idea which one may be the source of issue ?

lasofivec, 2019-06-11:
It looks to me is a problem with the new *.pxd files...
Can you give me the specs of the machines where it doesn't work ? (compiler version...)

Didou09, 2019-06-11:
**(base) [ DV226270 sirrah ~/ToFu_All/tofu_git/tofu (Issue61_Imas2tofuMoreRobust)] gcc --version**
gcc (GCC) 6.4.0
Copyright © 2017 Free Software Foundation, Inc.
Ce logiciel est libre; voir les sources pour les conditions de copie.  Il n'y a
AUCUNE GARANTIE, pas même pour la COMMERCIALISATION ni L'ADÉQUATION À UNE TÂCHE PARTICULIÈRE.

Didou09, 2019-06-11:
**(base) [ DV226270 sirrah ~/ToFu_All/tofu_git/tofu (Issue61_Imas2tofuMoreRobust)] ldd --version**
ldd (GNU libc) 2.12
Copyright © 2010 Free Software Foundation, Inc.
Ce logiciel est libre; voir les sources pour les conditions de
reproduction. AUCUNE garantie n'est donnée; tant pour des raisons
COMMERCIALES que pour RÉPONDRE À UN BESOIN PARTICULIER.
Écrits par Roland McGrath et Ulrich Drepper.

lasofivec, 2019-06-11:
and cython and python versions ?

Didou09, 2019-06-11:
**(base) [ DV226270 sirrah ~/ToFu_All/tofu_git/tofu (Issue61_Imas2tofuMoreRobust)] python --version**
Python 3.6.3 :: Anaconda custom (64-bit)
**(base) [ DV226270 sirrah ~/ToFu_All/tofu_git/tofu (Issue61_Imas2tofuMoreRobust)] cython --version**
Cython version 0.25.2

Didou09, 2019-06-11:
Looking at the [changelogs](https://cython.readthedocs.io/en/latest/src/changes.html#id1) of cython, there have been a few changes to the way it handles .pxd files.

From the description, none of them really seem to address directly our issue though, but since 0.25.2 is getting quite old, it might be a good idea to ask for an update to the admins (not sure it's possibl though).

Can you try to reproduce the compilation error if you downgrade to cython 0.25.2 ?

lasofivec, 2019-06-11:
I just pushed a new branch, do you mind trying that one:

> fixing-issue-89

lasofivec, 2019-06-11:
(there will probably still be two errors, but it should fix the first three)

Didou09, 2019-06-11:
(base) [ DV226270 sirrah ~/ToFu_All/tofu_git/tofu (fixing-issue-89)] python setup.py build_ext --inplace > debugCython.txt

Error compiling Cython file:
------------------------------------------------------------
...
# ==============================================================================
cdef inline bint inter_ray_aabb_box(const int[3] sign,
                                    const double[3] inv_direction,
                                    const double[6] bounds,
                                    const double[3] ds,
                                    bint countin=False,
                                   ^
------------------------------------------------------------

tofu/geom/_raytracing_tools.pxd:29:36: default values cannot be specified in pxd files, use ? or *

Error compiling Cython file:
------------------------------------------------------------
...
cdef inline bint inter_ray_aabb_box(const int[3] sign,
                                    const double[3] inv_direction,
                                    const double[6] bounds,
                                    const double[3] ds,
                                    bint countin=False,
                                    bint debug_plot=False) nogil:
                                   ^
------------------------------------------------------------

tofu/geom/_raytracing_tools.pxd:30:36: default values cannot be specified in pxd files, use ? or *

Error compiling Cython file:
------------------------------------------------------------
...
    bounds[3] = xmax
    bounds[4] = ymax
    bounds[5] = zmax
    return

cdef inline void coordshift_simple1d(double[3] pts, bint in_is_cartesian=True,
                                                   ^
------------------------------------------------------------

tofu/geom/_raytracing_tools.pxd:336:52: default values cannot be specified in pxd files, use ? or *

Error compiling Cython file:
------------------------------------------------------------
...
    bounds[4] = ymax
    bounds[5] = zmax
    return

cdef inline void coordshift_simple1d(double[3] pts, bint in_is_cartesian=True,
                                     double CrossRef=0., double cos_phi=0.,
                                    ^
------------------------------------------------------------

tofu/geom/_raytracing_tools.pxd:337:37: default values cannot be specified in pxd files, use ? or *

Error compiling Cython file:
------------------------------------------------------------
...
    bounds[4] = ymax
    bounds[5] = zmax
    return

cdef inline void coordshift_simple1d(double[3] pts, bint in_is_cartesian=True,
                                     double CrossRef=0., double cos_phi=0.,
                                                        ^
------------------------------------------------------------

tofu/geom/_raytracing_tools.pxd:337:57: default values cannot be specified in pxd files, use ? or *

Error compiling Cython file:
------------------------------------------------------------
...
    bounds[5] = zmax
    return

cdef inline void coordshift_simple1d(double[3] pts, bint in_is_cartesian=True,
                                     double CrossRef=0., double cos_phi=0.,
                                     double sin_phi=0.) nogil:
                                    ^
------------------------------------------------------------

tofu/geom/_raytracing_tools.pxd:338:37: default values cannot be specified in pxd files, use ? or *

Error compiling Cython file:
------------------------------------------------------------
...
                                      const double eps_vz, const double eps_a,
                                      const double eps_b, const double eps_pln,
                                      const bint is_in_struct,
                                      double[1] kpin_loc, double[1] kpout_loc,
                                      int[1] ind_loc, double[3] vperpin,
                                      bint debug_plot=False) nogil:
                                     ^
------------------------------------------------------------

tofu/geom/_raytracing_tools.pxd:723:38: default values cannot be specified in pxd files, use ? or *

Didou09, 2019-06-11:
There are many other errors below but let's focus on the first ones so far

Btw, I'm trying to export the output of 'python setup.py build_ext --inplace' to a txt file so I can send you the whole message in a more compact form but it does not seem to work, any tip ?

lasofivec, 2019-06-11:
I think I got the gist of the problem.
Just pushed on the same branch... I think all bugs should be fixed

Didou09, 2019-06-11:
(base) [ DV226270 sirrah ~/ToFu_All/tofu_git/tofu (fixing-issue-89)] python setup.py build_ext --inplace

Version for setup.py :  1.3.24-602-ge80f0ca

Using Cython !!!!!!!!!

Compiling tofu/geom/_GG03.pyx because it changed.
Compiling tofu/geom/_basic_geom_tools.pyx because it changed.
[1/2] Cythonizing tofu/geom/_GG03.pyx
[2/2] Cythonizing tofu/geom/_basic_geom_tools.pyx
/Applications/Anaconda/python36/lib/python3.6/site-packages/setuptools/dist.py:364: UserWarning: The version specified ('1.3.24-602-ge80f0ca') is an invalid version, this may not work as expected with newer versions of setuptools, pip, and PyPI. Please see PEP 440 for more details.
  "details." % self.metadata.version
running build_ext
building 'tofu.geom._GG03' extension
gcc -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/Applications/Anaconda/python36/lib/python3.6/site-packages/numpy/core/include -I/Applications/Anaconda/python36/include/python3.6m -c tofu/geom/_GG03.c -o build/temp.linux-x86_64-3.6/tofu/geom/_GG03.o -O0 -Wall -fopenmp
In file included from /Applications/Anaconda/python36/lib/python3.6/site-packages/numpy/core/include/numpy/ndarraytypes.h:1809:0,
                 from /Applications/Anaconda/python36/lib/python3.6/site-packages/numpy/core/include/numpy/ndarrayobject.h:18,
                 from /Applications/Anaconda/python36/lib/python3.6/site-packages/numpy/core/include/numpy/arrayobject.h:4,
                 from tofu/geom/_GG03.c:453:
/Applications/Anaconda/python36/lib/python3.6/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: attention : #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^~~~~~~
gcc -pthread -shared -B /Applications/Anaconda/python36/compiler_compat -L/Applications/Anaconda/python36/lib -Wl,-rpath=/Applications/Anaconda/python36/lib -Wl,--no-as-needed -Wl,--sysroot=/ build/temp.linux-x86_64-3.6/tofu/geom/_GG03.o -L/Applications/Anaconda/python36/lib -lpython3.6m -o build/lib.linux-x86_64-3.6/tofu/geom/_GG03.cpython-36m-x86_64-linux-gnu.so -fopenmp
building 'tofu.geom._basic_geom_tools' extension
gcc -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/Applications/Anaconda/python36/lib/python3.6/site-packages/numpy/core/include -I/Applications/Anaconda/python36/include/python3.6m -c tofu/geom/_basic_geom_tools.c -o build/temp.linux-x86_64-3.6/tofu/geom/_basic_geom_tools.o -O0 -Wall -fopenmp
gcc -pthread -shared -B /Applications/Anaconda/python36/compiler_compat -L/Applications/Anaconda/python36/lib -Wl,-rpath=/Applications/Anaconda/python36/lib -Wl,--no-as-needed -Wl,--sysroot=/ build/temp.linux-x86_64-3.6/tofu/geom/_basic_geom_tools.o -L/Applications/Anaconda/python36/lib -lpython3.6m -o build/lib.linux-x86_64-3.6/tofu/geom/_basic_geom_tools.cpython-36m-x86_64-linux-gnu.so -fopenmp
copying build/lib.linux-x86_64-3.6/tofu/geom/_GG03.cpython-36m-x86_64-linux-gnu.so -> tofu/geom
copying build/lib.linux-x86_64-3.6/tofu/geom/_basic_geom_tools.cpython-36m-x86_64-linux-gnu.so -> tofu/geom

Didou09, 2019-06-11:
Done   :-)

And yet another mission successfully completed by super-Laura !
Thanks very much,

Time for a quite debrief on Hangout ?

lasofivec, 2019-06-11:
Great !

closed 2019-06-11 by lasofivec

Didou09, 2019-06-11:
[image]

PR #91 (2019-06-11): Fixing issue 89
Fixes #89