ToFuProject/tofu #92: pip wheel : make sure wheel creation works out-of-the box

Didou09, 2019-06-13:
So far, wheel creation seems to work fine provided that the tests on openmp are commented out in the setup.py.

Find an elegant generic solution to that.

Method : create a dedicated branch from devel, named using the issue number

----------------------------------------------------------------------------
De : LE BOURDAIS Florian <Florian.LEBOURDAIS@cea.fr> 
Envoyé : jeudi 13 juin 2019 17:27
À : VEZINET Didier 226270 <Didier.VEZINET@cea.fr>
Objet : RE: Packaging Python

D’accord, merci pour l’info ! 

En fait, je génère des wheel à l’aide de la commande suivante :
Python setup.py bdist_wheel
Ca fonctionne tout seul normalement. D’ailleurs je viens de tester avec tofu et moyennant des imports manuels et la mise en commentaire du test openmp, j’ai réussi à obtenir un wheel :
 

Et en installant le wheel, j’ai bien réussi à importer tofu :
 

Donc a priori, le setup.py est bien configuré pour faire des wheel avec le code binaire généré par Cython !

C’est cool ça 

Je vais essayer de lancer les tests maintenant !
Florian

flothesof, 2019-06-13:
Some more details about the steps I followed before writing the above to @Didou09.

- create a new conda environment with Python 3.6 and pip
- clone the tofu git repo
- install Cython, numpy manually (because they are imported in the setup.py) using pip
- run `python setup.py bdist_wheel`

Results: the Cython code gets compiled and bundled inside a .whl file that I manage to install in the environment and that lets me import tofu.

I then ran the tests (10/24 failed).

lasofivec, 2019-06-17:
Is the "openmp test" is the script on the `setup.py` to test if `openmp` is installed or not ?
If so, what's the error ?

flothesof, 2020-05-07:
I think this can be closed now, since there has been a lot of improvements to setup and such. Also, this issue is quite vague, so nothing actionable here in my opinion. If generic problems arise with openmp, a new issue would be preferable I think.

@Didou09 @lasofivec ?

closed 2020-05-07 by Didou09