ToFuProject/tofu #93: Cython dependency : check whether dependecy can be relaxed in pre-compiled installs

Didou09, 2019-06-13:
If the user installs from a pre-compiled version (i.e.: conda or wheel), the cython dependency can probably be relaxed ?

Check if possible and check out standard / good practices in the community / other libraries.

Method : branch from devel with issue number

flothesof, 2019-06-13:
After looking around, I believe the way the setup.py is written is common practice.
The Cython docs mention the same lines that the ones we're using (see [here](http://docs.cython.org/en/latest/src/quickstart/build.html)).
Another package, choldate, does something similar to us: see [here](https://github.com/modusdatascience/choldate/blob/master/setup.py).

Didou09, 2019-06-13:
Thanks !

So we'll simply add a try / except statement at cython import time then, with an explicit warning message and test it out

flothesof, 2019-06-14:
I think this issue can be closed, in the sense that there is nothing to fix as of right now.
We can consider reopening later if needed.

Didou09, 2019-06-14:
Agreed

closed 2019-06-14 by Didou09