ToFuProject/tofu #95: nosetests not running on windows

flothesof, 2019-06-14:
I've been trying to get the tests running on Windows (Python 3.6 conda environment for tofu testing).
I've seen two problems so far.

# Path to data files used by tests
It appears that the way the path to the data files is handled breaks on my setup.
If I replace `_root = _here[:_here.rfind('/tofu')]` in tofu/geom/utils.py by `_path_testcases = os.path.join(os.path.split(_here)[0], 'inputs')`, tests manage to find the correct path to data files.

# Cython problems (?)
However, I then break on what I think are Cython related errors:
```python
Error
Traceback (most recent call last):
  File "C:\Anaconda3\envs\test_tofu\lib\unittest\case.py", line 59, in testPartExecutor
    yield
  File "C:\Anaconda3\envs\test_tofu\lib\unittest\case.py", line 605, in run
    testMethod()
  File "C:\Anaconda3\envs\test_tofu\lib\site-packages\nose\case.py", line 198, in runTest
    self.test(*self.arg)
  File "C:\Users\FL232714\Documents\Python workspace\Scripts\2019_conda packaging\tofu\tofu\tests\tests01_geom\tests01_GG.py", line 882, in test13_LOS_PInOut
    ves_type='Lin', test=True)
  File "tofu\geom\_GG03.pyx", line 1821, in tofu.geom._GG03.LOS_Calc_PInOut_VesStruct
ValueError: Buffer dtype mismatch, expected 'long' but got 'long long'
```
I'll investigate to see if I can go further (and then propose a PR).

Didou09, 2019-06-14:
Thanks a lot !

flothesof, 2019-06-14:
It seems that other people have had the same problem.

Here's a link to how they fixed it:
https://github.com/mdtraj/mdtraj/pull/1409/commits/46bf1431659ad80924ed11fc796338cb06845206

Looks like this needs some more investigation.

flothesof, 2019-06-14:
One of the functions that raises the error is `_Ves_Smesh_Lin_SubFromInd_cython`, found in `_GG03.pyx`.

I've tried to put a `long long` type for the `ind` variable in the signature of `def _Ves_Smesh_Lin_SubFromInd_cython(double[::1] XMinMax, double dL, double dX,
                                     double[:,::1] VPoly, np.ndarray[long,ndim=1] ind,
                                     double DIn=0., VIn=None, double margin=_VSMALL):`
but every time I run `python setup.py build_ext --inplace`, the code gets replaced by its original signature.
Any hints on why I can't manage to modify the cython?

lasofivec, 2019-06-14:
You have to change it in the _GG02.pyx file, the _GG03.pyx is only a copy of the GG02 to ensure that the python2 and python3 versions remain the same :)

lasofivec, 2019-06-14:
However, I think it should be better to force the type of all longs and/or ints at the beginning of the file (as done in the solution @flothesof linked to) is better than changing the signatures of the function (?)

flothesof, 2019-06-14:
Thanks for pointing out the copy mechanism. I'll see if I can get it to work following the solution `mdtraj` used and I'll report back here.

Didou09, 2019-06-14:
Yes, I had forgotten to explain, thanks Laura.
_GG03.pyx is a copy of _GG02.pyx produced when the setup.py is run, never edit _GG03.pyx, the changes would be lost at the next copy.

I agree that forcing all types at the beginning of the file is a good options
Thanks Flo, waiting for your changes

(if you create a local branch from devel, please also push it here on Github so we'll keep track of it and keep a common trace after it is deleted).

flothesof, 2019-06-14:
After trying the brute force solution (replacing all longs by int64), I am now able to run 85/114 tests on my test environment.
However, there are still errors related to this problem. I'm still trying to understand the problems. One observation I have is based on code where the variable is first declared and then initialized with a numpy array of integer datatype (from `_Ves_meshCross_FromD` here):
```
cdef np.ndarray[int64_t,ndim=1] ind
[...]
ind = np.empty((n1*n2,),dtype=int)
```
This leads to errors of the type `Buffer dtype mismatch, expected 'int64_t' but got 'long'`.

So brute force replacement is not enough...

lasofivec, 2019-06-14:
From this [SO answer](https://stackoverflow.com/questions/26877845/overflowerror-occurs-when-using-cython-with-a-large-int):

> Windows for both 32 and 64 bit uses 4 bytes (32 bits) for int and long, 8 bytes (64 bits) for long long. The difference between Win32 and Win64 is size of pointer (32 bits for Win32 and 64 bits for Win64). (See Data Type Ranges] from MSDN).
> 
> Linux uses another model: int is 32 bits for both linux-32 and linux-64, long long is always 64-bit.  long and pointers are vary: 32 bits on linux-32 and 64 bits on linux-64.

I think we could be using `long long` everywhere where we use `long`, I created a branch to test this: `long2longlong` as I dont have a Windows machine, I cannot try it, would you mind testing it for me ?

flothesof, 2019-07-24:
Hi all,
now retrying with Laura's *long2longlong* branch.
Here's the log of actions I performed.
- `conda create -n py36tofu python=3.6 pip cython numpy`
- `conda activate py36tofu`
- `git clone https://github.com/ToFuProject/tofu.git`
- `git checkout long2longlong`
- `pip install -e .[test]` with some modifications of setup.py (the openmp check on line 70 fails, so I had to avoid it and write `not_openmp_installed = True`)
- `nosetests` from the current directory and having fixed `_path_testcases = os.path.join(os.path.split(_here)[0], 'inputs')`

I get the following 
```
----------------------------------------------------------------------
Ran 115 tests in 55.602s

FAILED (errors=22, failures=8)

```
I'm attaching the nosetest log to the issue for further analysis.
[test_report.txt](https://github.com/ToFuProject/tofu/files/3425600/test_report.txt)

flothesof, 2019-07-24:
I'm now down to 3 errors and 6 failures. Submitting a pull request with the changes.

lasofivec, 2019-07-24:
> I believe some of the errors are related to the longlong changes.

It seems that way, except for a few of the type:

```
======================================================================
ERROR: tofu.tests.tests02_data.tests03_core.Test01_DataCam12D.test06_set_dtreat_indch
----------------------------------------------------------------------
Traceback (most recent call last):
  File "c:\anaconda3\envs\py36tofu\lib\site-packages\nose\case.py", line 198, in runTest
    self.test(*self.arg)
  File "C:\Users\FL232714\Desktop\tofu_issue95\tofu\tofu\tests\tests02_data\tests03_core.py", line 280, in test06_set_dtreat_indch
    oo.set_dtreat_indch(indch = range(0,10))
  File "C:\Users\FL232714\Desktop\tofu_issue95\tofu\tofu\data\_core.py", line 942, in set_dtreat_indch
    indch = _format_ind(indch, n=self._ddataRef['nch'])
  File "C:\Users\FL232714\Desktop\tofu_issue95\tofu\tofu\data\_core.py", line 70, in _format_ind
    raise Exception(msg)
Exception: Index must be a int, or an iterable of bool or int !
```

Did they go away while solving the reste of the errors ?
And how are you fixing the errors ?

Will try the branch in a couple of minutes.

flothesof, 2019-07-24:
I basically changed the calls until I had long long's everywhere in the .pyx file. It was basically about forcing the right type of integer where conversions were happening.

lasofivec, 2019-07-30:
Passing on:
- two different Ubuntu 16.04.6 LTS;
- Ubuntu 18.10
- Mac OSX 10.13

lasofivec, 2019-08-20:
@flothesof I'm trying to test ToFu on GitHub, how did you installed it ?
Are you using Cygwin + conda ?
I'm blocked trying to install Scipy on Cygwin, and from what I've read, it seems easier to use Conda + Cygwin. I was curious on what you are doing in order to make it work

flothesof, 2019-08-21:
Hey Laura,
I've described my setup on a windows machine 4 comments ago. I'm using conda to create an environment for Tofu and then installing from source, basically.

lasofivec, 2019-08-21:
Yes, thanks.
But I guess my question is more abot the terminal. Are you using Cygwin or Git bash or Anaconda's prompt ?

flothesof, 2019-08-21:
I'm using the Anaconda prompt.

flothesof, 2019-09-16:
Follow-up to this issue in #197: abandon long2longlong and just fix the couple of issues I had on Windows due to platform differences.

lasofivec, 2019-09-16:
fine by me :)

lasofivec, 2019-09-17:
we can erase the branch long2longlong then, right ?

flothesof, 2019-09-18:
Yes, we can erase that branch. I think the brute-force approach for long2longlong is not very convenient.

flothesof, 2019-11-27:
Closing since tests running on Windows now.

closed 2019-11-27 by flothesof