ToFuProject/tofu #102: Solid Angles: improve _GG.LOS_isVis_PtFromPts_VesStruct (flexibilty, kwdargs and multiple points)

Didou09, 2019-06-19:
Function _GG.LOS_isVis_PtFromPts_VesStruct() needs to be modified along the following lines:

* should take not 1 but N pts as input (and return a N,M) array of boolean indices
* k should be an optional keyword arguments (computed if not provided)
* VPoly and VIn should be keyword arguments

This way teh fonction can easily be called for a large number of points from a config instance, and the inputs can be provided by an appropriate dictionnary produced by a specific Config class method.

lasofivec, 2019-06-20:
I'm debuggin right now, and I am having a question for the 'k' and 'vis' arguments, as there are no comments I rather asked: What are the roles of `k` and `vis` ?

    indok = ~(np.isnan(k) | np.isnan(kPOut))
    if vis:
        ind = np.zeros((npts,),dtype=bool)
        ind[indok] = k[indok] < kPOut[indok]
    else:
        ind = np.ones((npts,),dtype=bool)
        ind[indok] = k[indok] > kPOut[indok]
    return ind

It seems that if we have a certain result `ind` with `vis=True`, then we will get `~ind` with `vis=False` and that `k` is a `kin` (a.k.a. `kmin`), is this exact ?

Didou09, 2019-06-25:
Hi (sorry, was away the last days),

k is the distance from pts to Points (sometimes already calculated and provided by the user, if not it must be calculated).
vis is a keyword argument to indicate what we want as an output.

In principle, if we have:
```python
ind0 = LOS_isVis_PtFromPts_VesStruct(..., vis=False, ...)
ind1 = LOS_isVis_PtFromPts_VesStruct(..., vis=True, ...)
```

We should have ind0 = ~ind1,
But in practice there may be some points for which the answer is not known because the intermediate function Calc_LOS_PInOut_Tor return NaN.
For those points, they are considered as False if vis=True, and False again if vis=False (so the function on ly returns True when the result is known, reliable and clear).

There might be a more clever way to do that, though.

Didou09, 2019-06-25:
Also, it would faster to call an intermediate function (Calc_LOS_PInOut_Tor) that only computes k (and not the 3 coordinates of each points)

PR #104 (2019-06-26): Issue102
Had to make some major changes to not re write code and to not slow it down.
Created function that creates kwdargs

Fixes #102
PR #106 (2019-07-05): Issue102
## Geometry changes

- Algorithm `_ves_is_inside` that checks if a list of point is inside a vessel (or structure) : updated, properly cythonized
- From ` LOS_Calc_PInOut_VesStruct` extracted core of the function as it is needed in other functions, to avoid repetitions
- `LOS_isVis_PtFromPts_VesStruct` checks if a point is visible with respect to a list of points and taking into account a vessel (with structures, limits, etc.) : function cythonized properly, speed up ~50%
- `LOS_areVis_PtsFromPts_VesStruct` equivalent to last function but input is a list of points
- `Dust_calc_SolidAn [... cut]
closed 2019-07-05 by lasofivec