ToFuProject/tofu #108: calc_LOS_signal2 : choice of strategy and mutualization of sampling

Didou09, 2019-07-05:
In _GG:

- LOS_calc_signal() should be replaced by LOS_calc_signal2()
- LOS_calc_signal2() has been drafted (branch from Issue61_Imas2TofuMoreRobust)

- It introduces: a choice between three strategies (minimize user's function calls by using vectorized routines for small number of LOS, minimize memory usage by looping for large number of LOS)

- It needs both the existing LOS_get_sample (for minimize = calls) and a single-LOS version of sampling (for minimize = memory)

- Making that single-LOS version of LOS_get_sample() may require revizing a bit the _sampling_tools.pxd to introduce inlined single-LOS unparallelized versions of all sampling methods.

This update will make computation of synthetic signal faster / more efficient (especially for large number of LOS) and more robust vs user's functions (if ill-defined, we can minimize the number of calls).

It will also make synthetic signal computation from Plasma2D objects easier and more natural

Didou09, 2019-07-05:
@lasofivec , now branch Issue61_Imas2TofuMoreRobust is up-to-date with devel (thanks to PR #114 )

You can have a look at the drafted solution in _GG02.pyx (LOS_calc_signal2) from this branch

lasofivec, 2019-07-24:
Branch `Issue108...` is failing because of some changes in `Issue61_Imas2TofuMoreRobust` with `python2`
See [this travis fail report](https://travis-ci.org/ToFuProject/tofu/jobs/563091008)

```
INFO conda_build.build:bundle_conda(891): Packaging tofu-1.4.0.341.g392e0c0-py27h39e3cac_0
  File "lib/python2.7/site-packages/tofu-1.4.0_319_g7fbfab5-py2.7-linux-x86_64.egg/tofu/data/_core.py", line 1889
    data = chronos[:,modes] @ (s[modes,None] * topos[modes,:])
                            ^
SyntaxError: invalid syntax
```

lasofivec, 2019-07-25:
On `Python3` there are other fails that come from the original branch, the only one that comes from `LOS_calc_signal` is the following.

```
======================================================================
FAIL: tofu.tests.tests01_geom.tests03_core.Test03_Rays.test10_calc_signal
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/travis/miniconda/lib/python3.6/site-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/home/travis/miniconda/lib/python3.6/site-packages/tofu/tests/tests01_geom/tests03_core.py", line 870, in test10_calc_signal
    fs=(12,6), connect=connect)
  File "/home/travis/miniconda/lib/python3.6/site-packages/tofu/geom/_core.py", line 4590, in calc_signal
    connect=connect)
  File "/home/travis/miniconda/lib/python3.6/site-packages/tofu/geom/_core.py", line 4510, in _calc_signal_postformat
    osig = tfd.DataCam2D(**kwdargs)
  File "/home/travis/miniconda/lib/python3.6/site-packages/tofu/data/_core.py", line 157, in __init__
    super(DataAbstract,self).__init__(**kwdargs)
  File "/home/travis/miniconda/lib/python3.6/site-packages/tofu/utils.py", line 973, in __init__
    self._init(**kwdargs)
  File "/home/travis/miniconda/lib/python3.6/site-packages/tofu/data/_core.py", line 610, in _init
    self._set_ddataRef(**kwddataRef)
  File "/home/travis/miniconda/lib/python3.6/site-packages/tofu/data/_core.py", line 631, in _set_ddataRef
    lout = self._checkformat_inputs_ddataRef(**kwdargs)
  File "/home/travis/miniconda/lib/python3.6/site-packages/tofu/data/_core.py", line 274, in _checkformat_inputs_ddataRef
    assert ndim in [2,3]
AssertionError
```

However, this issue come from the method `_calc_signal_postformat` where we are trying to build a camera to 2D from the integrated signal, however the signal is 1D. I rather not change this myself and pass it on to you. I will make a pull request and close this issue.

PR #118 (2019-07-25): Issue108 calc los signal
This answers Issue #108 

> In _GG:
> 
> - LOS_calc_signal() should be replaced by LOS_calc_signal2()
> - LOS_calc_signal2() has been drafted (branch from Issue61_Imas2TofuMoreRobust)
> 
> - It introduces: a choice between three strategies (minimize user's function calls by using vectorized routines for small number of LOS, minimize memory usage by looping for large number of LOS)
> 
> - It needs both the existing LOS_get_sample (for minimize = calls) and a single-LOS version of sampling (for minimize = memory)
> - Making that single-LOS version of LOS_get_sample() may [... cut]
closed 2019-07-30 by lasofivec