ToFuProject/tofu #121: No module Equimap

lasofivec, 2019-07-30:
It looks like there is a file missing (probably `equimap.py` ?):

```
======================================================================
ERROR: Failure: ModuleNotFoundError (No module named 'equimap')
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/mendoza/anaconda3/envs/tofu3/lib/python3.7/site-packages/nose/failure.py", line 39, in runTest
    raise self.exc_val.with_traceback(self.tb)
  File "/Users/mendoza/anaconda3/envs/tofu3/lib/python3.7/site-packages/nose/loader.py", line 417, in loadTestsFromName
    addr.filename, addr.module)
  File "/Users/mendoza/anaconda3/envs/tofu3/lib/python3.7/site-packages/nose/importer.py", line 47, in importFromPath
    return self.importFromDir(dir_path, fqname)
  File "/Users/mendoza/anaconda3/envs/tofu3/lib/python3.7/site-packages/nose/importer.py", line 94, in importFromDir
    mod = load_module(part_fqname, fh, filename, desc)
  File "/Users/mendoza/anaconda3/envs/tofu3/lib/python3.7/imp.py", line 234, in load_module
    return load_source(name, filename, file)
  File "/Users/mendoza/anaconda3/envs/tofu3/lib/python3.7/imp.py", line 171, in load_source
    module = _load(spec)
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/Users/mendoza/tofu/tofu/mag/test_equimap.py", line 19, in <module>
    import equimap
ModuleNotFoundError: No module named 'equimap'

```

Didou09, 2019-08-23:
@lasofivec can you confirm that this was due to a naming issue of some files containing the term 'test' and that nosetest was trying to run ? 
I am pretty sure it was the issue and you somehow solved it somewhere ?

lasofivec, 2019-08-26:
yes, it was solved by PR #154 and #146

closed 2019-08-26 by lasofivec