ToFuProject/tofu #122: Error while using local C++ pointer in parallel loop

lasofivec, 2019-07-30:
Error found in a machine with the following specifications

> Linux 64 bits (CentOS 6.9)
> Cython:  0.25.2 
> gcc :    6.4.0
> g++ :    6.4.0 (but not used a priori)
> GLIBC :  2.12 

Found the following error.

```
tofu/geom/_vignetting_tools.cpp:3034:55: erreur : invalid conversion from «int» to «double*» [-fpermissive]
                       double * __pyx_parallel_temp0 = 1;
                                                       ^
tofu/geom/_vignetting_tools.cpp:3036:52: erreur : invalid conversion from «int» to « int*» [-fpermissive]
                       int * __pyx_parallel_temp2 = 1;
                                                    ^
error: command 'gcc' failed with exit status 1
 
(base) [ DV226270 sirrah ~/ToFu_All/tofu_git/tofu (Issue61_Imas2tofuMoreRobust)] python setup.py build_ext --inplace

Version for setup.py :  1.4.0-338-g841d1f8

Compiling tofu/geom/_vignetting_tools.pyx because it changed.
[1/1] Cythonizing tofu/geom/_vignetting_tools.pyx

Error compiling Cython file:
------------------------------------------------------------
...
    with nogil, parallel(num_threads=num_threads):
        diff = NULL
        lref = NULL
        for ivign in prange(nvign):
            nvert = lnvert[ivign]
            diff = <double*>malloc(3*nvert*sizeof(double))
                ^
```

Compilation line (produced by setup.py):

`gcc -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/Applications/Anaconda/python36/lib/python3.6/site-packages/numpy/core/include -I/Applications/Anaconda/python36/include/python3.6m -c tofu/geom/_vignetting_tools.cpp -o build/temp.linux-x86_64-3.6/tofu/geom/_vignetting_tools.o -O0 -Wall -fopenmp`

Didou09, 2019-08-06:
After merging from 122_Pointer_Cpp to Issue61_Imas2TofuMoreRobust, on the IRFM servers I still get the following error at compilation:

`[ DV226270 spica ~/ToFu_All/tofu_git/tofu (Issue61_Imas2tofuMoreRobust)] python setup.py build_ext --inplace

Version for setup.py :  1.4.0-385-g4d0621f

your setuptools is too old (<12)
setuptools_scm functionality is degraded
/Applications/Anaconda/python36/lib/python3.6/site-packages/setuptools/dist.py:481: UserWarning: The version specified ('1.4.0-385-g4d0621f') is an invalid version, this may not work as expected with newer versions of setuptools, pip, and PyPI. Please see PEP 440 for more details.
  "details." % self.metadata.version
running build_ext
building 'tofu.geom._vignetting_tools' extension
gcc -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/Applications/Anaconda/python36/lib/python3.6/site-packages/numpy/core/include -I/Applications/Anaconda/python36/include/python3.6m -c tofu/geom/_vignetting_tools.cpp -o build/temp.linux-x86_64-3.6/tofu/geom/_vignetting_tools.o -O0 -Wall -fopenmp
cc1plus: warning: command line option \u2018-Wstrict-prototypes\u2019 is valid for C/ObjC but not for C++
tofu/geom/_vignetting_tools.cpp: In function \u2018int __pyx_f_4tofu_4geom_17_vignetting_tools_triangulate_polys(double**, long int*, int, long int**, int)\u2019:
tofu/geom/_vignetting_tools.cpp:3053:55: error: invalid conversion from \u2018int\u2019 to \u2018double*\u2019 [-fpermissive]
                       double * __pyx_parallel_temp0 = 1;
                                                       ^
tofu/geom/_vignetting_tools.cpp:3055:52: error: invalid conversion from \u2018int\u2019 to \u2018int*\u2019 [-fpermissive]
                       int * __pyx_parallel_temp2 = 1;
                                                    ^
error: command 'gcc' failed with exit status 1
`

Didou09, 2019-08-08:
Temporary fix : un-parallelize loop (cf. PR #127)

lasofivec, 2019-08-19:
Looks like this is only happening with numpy version = 1.15.4

lasofivec, 2019-08-19:
We are going to update requirements to

> cython version >= 0.26.1

lasofivec, 2019-08-19:
Fixed in branch Issue122_Pointer_Cpp

Didou09, 2019-08-27:
Fixed in devel by PR #159

Didou09, 2019-09-05:
Closed in 1.4.1

PR #123 (2019-08-06): Issue122 pointer cpp
Trying to fix error in CentOS.
Now working on (local) Docker. Tried to add docket to travis but I couldn't

Fixes #122
PR #133 (2019-08-19): Update to Cython >= 0.26 for bug fixes
Cython version 0.25 had some bugs that were causing the fails reported on Issue #122 
Now, *ToFu* requires version 0.26 or newer.
Made changes for conda build recipe, pip install and README file.

Fixes #122
PR #159 (2019-08-27): [Issue122]
# Version of Cython required is now > 0.26

Due to a bug in ``Cython``, ``ToFu`` requires now a version >= 0.26. Now the CEA servers are up to date so we can use resolve Issue #122 and set the pointer in the parallel loop as first intended.

Fixes #122
closed 2019-09-05 by Didou09