ToFuProject/tofu #128: Matplotlib error creating camera with CamLoS1D

jdhare, 2019-08-14:
For one specific camera, I get the following error:  

```
D=array([[ 4.09044068,  0.        , -3.34447391],
       [ 4.09472034,  0.        , -3.34173696],
       [ 4.099     ,  0.        , -3.339     ],
       [ 4.10327966,  0.        , -3.33626304],
       [ 4.10755932,  0.        , -3.33352609]])

u=array([[ 0.04536299,  0.        , -0.99897057],
       [ 0.30236989,  0.        , -0.95319067],
       [ 0.53877079,  0.        , -0.8424524 ],
       [ 0.73845534,  0.        , -0.67430239],
       [ 0.88781539,  0.        , -0.46019978]])

config = tf.geom.utils.create_config('A2')
cam = t.geom.CamLOS1D(dgeom=(D, u), config=config, Exp='ITER',
                                   Diag='Bolometer',Name='1')
```
```
File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/geom/_core.py", line 2497, in __init__
    super(Rays,self).__init__(**kwdargs)
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/utils.py", line 719, in __init__
    self._init(**kwdargs)
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/geom/_core.py", line 2867, in _init
    self._set_dgeom(sino=True, **kwdgeom)
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/geom/_core.py", line 3319, in _set_dgeom
    self.compute_dgeom(extra=extra)
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/geom/_core.py", line 3229, in compute_dgeom
    Nstep=50)
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/geom/_plot.py", line 717, in _LOS_calc_InOutPolProj_Debug
    fs=fs, wintit=wintit, draw=False)
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/geom/_core.py", line 2331, in plot
    wintit=wintit, tit=tit, Test=Test)
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/geom/_plot.py", line 196, in Struct_plot
    draw=False, **dplot[proj])
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/geom/_plot.py", line 471, in _Plot_3D_plt_Ves
    ax = _def.Plot_3D_plt_Tor_DefAxes(fs=fs, wintit=wintit)
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/geom/_def.py", line 152, in Plot_3D_plt_Tor_DefAxes
    ax.set_aspect(aspect="equal", adjustable='datalim')
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/matplotlib/axes/_base.py", line 1281, in set_aspect
    'It is not currently possible to manually set the aspect '
NotImplementedError: It is not currently possible to manually set the aspect on 3D axes
```
A Maplotlib 3D plot opens with nothing in it. Other cameras are created correctly.

Didou09, 2019-08-19:
Thanks for reporting,

2 things:

* On camera creation, if a 3D plot tries to open, it means tofu detected some LOS for which no solution exists (i.e.: for which it can detect no intersection with the vacuum vessel or any other structure). The 3D plot is simply part of the debugging process : when a LOS with no solution is detected, tofu tries to plot it in 3D so you can better visualize where the problem might come from).
Since matplotlib is not always flawless with 3D plots, it may happen, depending on your install that the plotting attempt fails for some reason.
A mid-term fix will be to complement that 3D plot by more robust 2D plots => Issue #131 opened (low-priority)

* Even though the plot did not succeed, it still indicates that there is probably an issue with the provided coordinates (i.e. one of the LOS is probably not valid and not intersecting the vessel).
Indeed, if you consider the first LOS, the X coordinate of its vector u is 10 times smaller than all others, and if we try multiplying it by 10, the following works fine:

```
D=array([[ 4.09044068,  0.        , -3.34447391],
       [ 4.09472034,  0.        , -3.34173696],
       [ 4.099     ,  0.        , -3.339     ],
       [ 4.10327966,  0.        , -3.33626304],
       [ 4.10755932,  0.        , -3.33352609]])

u=array([[ 0.4536299,  0.        , -0.99897057],
       [ 0.30236989,  0.        , -0.95319067],
       [ 0.53877079,  0.        , -0.8424524 ],
       [ 0.73845534,  0.        , -0.67430239],
       [ 0.88781539,  0.        , -0.46019978]])

config = tf.geom.utils.create_config('A2')
cam = t.geom.CamLOS1D(dgeom=(D, u), config=config, Exp='ITER',
                                   Diag='Bolometer',Name='1')```

Please, double-check those coordinates and report back on whether there was a coordinate issue.
Issue #131 will be addressed any way.

jdhare, 2019-08-20:
Hi Didier,

You are correct that the LoS was invalid. We had an error in our spreadsheet which specifies the angle between adjacent LoS and the first LoS was outside the vessel. In 1.4.1 the error message/warning message is much more helpful and specified which LoS was not inside the vessel, so that is useful!

Didou09, 2019-08-20:
Thanks,

I'm marking this issue as closed in the next release then.
Making the plots more robust will be addressed later in Issue #131

Didou09, 2019-09-05:
Closed in 1.4.1

closed 2019-09-05 by Didou09