ToFuProject/tofu #129: Camera plot won't accept array of AxesSubplots

jdhare, 2019-08-14:
A common method of generating a set of aces using matplotlib is with a call to plt.subplots. However, this returned list of axes is not accepted by the plot function in tofu. I think this is overzealous type checking.
```
fig, ax=plt.subplots(1,2)
cam.plot(ax) # cam is tofu.geom._core.CamLOS1D
Traceback (most recent call last):
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/IPython/core/interactiveshell.py", line 3326, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-90-cda99779acfe>", line 1, in <module>
    cam.plot(ax)
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/geom/_core.py", line 4323, in plot
    fs=fs, wintit=wintit, draw=draw, Test=Test)
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/geom/_plot.py", line 809, in Rays_plot
    Lax, C0, C1, C2 = _check_Lax(Lax, n=2)
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/geom/_plot.py", line 51, in _check_Lax
    assert C0 or C1 or C2, "Arg Lax must be an Axes or a list/tuple of such !"
AssertionError: Arg Lax must be an Axes or a list/tuple of such !
```

Didou09, 2019-08-19:
Agreed, even though:

* The problem does not come from the fact that you are providing subplots instead of regular axes, but rather from the fact you are providing it as a np.array of 2 subplots. The type checking routine handles lists and tuples of axes / subplots, but we did not think about arrays of such. 
It should however properly handle such cases because as you underline, an array of subplot as provided by command plt.subplots() is a common use-case, so the routine will be made more robust and able to handle such cases, from version 1.4.1, thanks for the feedback.

Didou09, 2019-08-19:
Fixed in devel by PR #132

Didou09, 2019-09-05:
Closed in 1.4.1

closed 2019-09-05 by Didou09