ToFuProject/tofu #130: Adding cameras

jdhare, 2019-08-14:
I think that Didier told me that it possible to add tofu.geom.CamLOS1D objects together. I can't seem to get this work in tofu 1.4.0 - is it a feature for the next release? It would be very useful for building our pinhole cameras, because they are constructed from several subcameras.

Didou09, 2019-08-19:
This feature is already available in the development version (since PR #100 ) and will be available in the next release (1.4.1)

Example use case:
============

```
import numpy as np
import tofu as tf

D0 = np.array([[ 4.09044068,  0.        , -3.34447391],
                        [ 4.09472034,  0.        , -3.34173696]])
D1 = np.array([[ 4.10327966,  0.        , -3.33626304],
                        [ 4.10755932,  0.        , -3.33352609]])

u0 = np.array([[ 0.04536299,  0.        , -0.99897057],
                        [ 0.30236989,  0.        , -0.95319067]])
u1 = np.array([[ 0.73845534,  0.        , -0.67430239],
                        [ 0.88781539,  0.        , -0.46019978]])

config = tf.geom.utils.create_config('A2')
cam0 = tf.geom.CamLOS1D(dgeom=(D0, u0), config=config, Exp='ITER',
                                              Diag='Bolometer', Name='0')
cam1 = tf.geom.CamLOS1D(dgeom=(D1, u1), config=config, Exp='ITER',
                                              Diag='Bolometer', Name='1')

cam = cam0 + cam1
```

jdhare, 2019-08-20:
Excellent, thank you.

Didou09, 2019-09-05:
Closed in 1.4.1

closed 2019-09-05 by Didou09