ToFuProject/tofu #147: Error when pandas not installed

lasofivec, 2019-08-23:
In ``tofu/geom/_core.py`` in the function ``get_summary`` there should be a test before using ``pandas`` function.
Error

```
======================================================================
ERROR: tofu.tests.tests01_geom.tests03_core.Test02_Config.test09_get_summary
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Home/LM257054/.conda/envs/tofu3/lib/python3.6/site-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/Home/LM257054/tofu/tofu/tests/tests01_geom/tests03_core.py", line 541, in test09_get_summary
    self.dobj[typ].get_summary()
  File "/Home/LM257054/tofu/tofu/geom/_core.py", line 2392, in get_summary
    df = pd.DataFrame(data, columns=col)
NameError: name 'pd' is not defined

```

Didou09, 2019-08-23:
Fixed (devel) by PR #152

Didou09, 2019-09-05:
Closed in 1.4.1

closed 2019-09-05 by Didou09