ToFuProject/tofu #160: _GG.LOS_get_sample(): bug in mode=rel and slowdown

Didou09, 2019-08-27:
Minimum working example:
==================

Considering 2 LOS, we want to sample them from in the intervals k=[1,2] and [10,20] respectively. All k output should be in these intervals, but:
dL = 0.25
DL = np.array([[1.,10.],[2.,20.]])
k = _GG.LOS_get_sample(2, dL, DL, dmethod='rel', method='sum')[0]
print(k)
[80., 52.8, 576., 52.8, 11.25....]

Problem appears when:
* dL unique + rel + sum    => Error !!  (k too large)
* dL unique + rel + simps  => ok
* dL unique + rel + romb  => ok
* dL unique + abs + sum  => ok
* dL unique + abs + simps  => ok
* dL unique + abs + romb  => ok

* dL unique + rel + sum  => Error !  (k too small)
* dL unique + rel + simps  => Error ! (k too small)
* dL unique + rel + romb  => Error!   (k too small)
* dL unique + abs + sum  => ok
* dL unique + abs + simps  => ok

When k is too small, it seems to be due to line 733 in tofu/geom/_sampling_tools.pyx (for the case 'simps', but I guess it is similar for others): the local resolution should not be 1/num_raf but seg_length / num_raf like in line 782.
The 'relative' aspect is already taken into account in line 730 (vs 779).

Please debug and double-check all results

For the case k too large... no time to check, please investigate

Also, optimization:
============
In almost all routines in _sampling_tools.pyx, there is a if test (if ii == 0) inside a for loop.
Like in previous case, please take the test out (run first iteration separately) to speed up computation

Maintenance:
=========
* It looks like many of the lins inside the for loops have a lot (almost everything) in common, please consider inlining for easier maintenance and better readability.
* Functions like simps_left_rule_abs_single (line 750) and simps_left_rule_rel_single (line 703) seem to be exactly the same . If so, consider unifying / deleting duplicate for better readability and maintenance

lasofivec, 2019-08-27:
in which branch ?? devel ?

lasofivec, 2019-08-27:
There is a bug for ``k = _GG.LOS_get_sample(2, dL, DL, dmethod='rel', method='sum')[0]``
Actually found it in another branch that I am working on right now, but I will submit the bug fix rn.

However, for the other methods, I haven't found any issues. I added a unit test:

> import tofu.geom._GG as _GG
> import numpy as np
> 
> dmeths = ['rel', 'abs']
> qmeths = ['simps', 'romb', 'sum']
> 
> dL = 0.25
> DL = np.array([[1.,10.],[2.,20.]])
> 
> for dm in dmeths:
>     for qm in qmeths:
>         print("============ for: ", dm, " -", qm, "=======")
>         out = _GG.LOS_get_sample(2, dL, DL, dmethod=dm, method=qm)
>         k = out[0]
>         lind = out[2]
>         print(" k =", k)
>         print("lind =", lind)
>         assert np.all(k[:lind[0]] >= DL[0][0])
>         assert np.all(k[:lind[0]] <= DL[1][0])
>         assert np.all(k[lind[0]:] >= DL[0][1])
>         assert np.all(k[lind[0]:] <= DL[1][1])
>         print("================= OK =======================")

lasofivec, 2019-08-27:
As for the other two points:

- You already pointed out the ``if`` statement and I have taken them out, the PR is coming... (branch ``Issue142_calc_signal_perf`` the "perf" stands for performance as I am optimizing different things for ``calc_signal`` and ``get_sample`` probably should have done small branches, sorry)
- Repetition of code: ok going to apply this in the same branch ``Issue142_calc_signal_perf`` as PR is coming really soon

lasofivec, 2019-08-27:
By the way, the result to my code two comments ago is the following.

> ============ for:  rel  - simps =======
>  k for 1st LOS = [1.   1.25 1.5  1.75 2.  ]
>  k for 1st LOS = [10.  12.5 15.  17.5 20. ]
> lind = [5]
> ================= OK =======================
> ============ for:  rel  - romb =======
>  k for 1st LOS = [1.   1.25 1.5  1.75 2.  ]
>  k for 1st LOS = [10.  12.5 15.  17.5 20. ]
> lind = [5]
> ================= OK =======================
> ============ for:  rel  - sum =======
>  k for 1st LOS = [1.125 1.375 1.625 1.875]
>  k for 1st LOS = [11.25 13.75 16.25 18.75]
> lind = [4]
> ================= OK =======================
> ============ for:  abs  - simps =======
>  k for 1st LOS = [1.   1.25 1.5  1.75 2.  ]
>  k for 1st LOS = [10.   10.25 10.5  10.75 11.   11.25 11.5  11.75 12.   12.25 12.5  12.75
>  13.   13.25 13.5  13.75 14.   14.25 14.5  14.75 15.   15.25 15.5  15.75
>  16.   16.25 16.5  16.75 17.   17.25 17.5  17.75 18.   18.25 18.5  18.75
>  19.   19.25 19.5  19.75 20.  ]
> lind = [5]
> ================= OK =======================
> ============ for:  abs  - romb =======
>  k for 1st LOS = [1.   1.25 1.5  1.75 2.  ]
>  k for 1st LOS = [10.      10.15625 10.3125  10.46875 10.625   10.78125 10.9375  11.09375
>  11.25    11.40625 11.5625  11.71875 11.875   12.03125 12.1875  12.34375
>  12.5     12.65625 12.8125  12.96875 13.125   13.28125 13.4375  13.59375
>  13.75    13.90625 14.0625  14.21875 14.375   14.53125 14.6875  14.84375
>  15.      15.15625 15.3125  15.46875 15.625   15.78125 15.9375  16.09375
>  16.25    16.40625 16.5625  16.71875 16.875   17.03125 17.1875  17.34375
>  17.5     17.65625 17.8125  17.96875 18.125   18.28125 18.4375  18.59375
>  18.75    18.90625 19.0625  19.21875 19.375   19.53125 19.6875  19.84375
>  20.     ]
> lind = [5]
> ================= OK =======================
> ============ for:  abs  - sum =======
>  k for 1st LOS = [1.125 1.375 1.625 1.875]
>  k for 1st LOS = [10.125 10.375 10.625 10.875 11.125 11.375 11.625 11.875 12.125 12.375
>  12.625 12.875 13.125 13.375 13.625 13.875 14.125 14.375 14.625 14.875
>  15.125 15.375 15.625 15.875 16.125 16.375 16.625 16.875 17.125 17.375
>  17.625 17.875 18.125 18.375 18.625 18.875 19.125 19.375 19.625 19.875]
> lind = [4]
> ================= OK =======================
>

Didou09, 2019-08-27:
Sorry I realize my first comment was inaccurate:

The first 6 cases are was dL unique (dL = 0.25)
The last 6 cases are for dL array (dL = np.r_[0.25, 0.25])

Sorry, due to copy-paste...

May I ask you test also the 6 cases with dL as an array, check if there are errors (there should be the 3 ponted out), fix the bug, double-check and include in unit tests ?

Thanks !

Didou09, 2019-08-27:
The part with dL unique is indeed solved by PR #161 

The part with dL array remains to be debugged

Didou09, 2019-08-28:
Last bug (dL array) fixed in devel by PR #162

Didou09, 2019-09-05:
Closed in 1.4.1

closed 2019-09-05 by Didou09