ToFuProject/tofu #174: update of documentation

lasofivec, 2019-09-05:
Since we have a new user,  @munechika-koyo, updating the documentation is, in my opinion the next *urgent* change for **tofu**.
I created a branch ``docupdate`` for this issue.

flothesof, 2019-09-10:
Hi @lasofivec 
It's a great idea to improve the docs. The figures you've added area are also very nice.
I have two suggestions regarding this improvement:
- build the documentation automatically via travis and upload it 
- use [sphinx-gallery](https://sphinx-gallery.github.io/) to generate a gallery of examples
Do you think this would be doable? I could probably help with the setup of sphinx-gallery (since I've done this before).

lasofivec, 2019-09-10:
Hi @flothesof thank's for the feedback. Regarding your points

> - build the documentation automatically via travis and upload it

Yes, this is for sure the right way of handling the doc. However, the configuration of this can only be done by @Didou09 (currently in the `Settings` of the project via GitHub). And I think he won't be able to check this for a few more days.

> - use sphinx-gallery to generate a gallery of examples

I didn't know this, but it looks really nice. If you want to do it, I'll be more than happy :)
We added a few things to the README, and the wikipedia as a quick-fix solution. Hopefully in the near future everything will be in a single place and up to date !

flothesof, 2019-09-13:
A short note as I progress. It seems that sphinx is not listed as a dependency in setup.py.

As a user, this makes perfect sense. 
As a developer, it doesn't.

I suggest to do the following. Add an extra 'dev' option in the dependencies that installs both the testing infrastructure and everything needed to locally generate the docs. This would replace the current 'test' and 'dev' options.

I'm going to add this my sphinx_gallery branch.

lasofivec, 2019-09-13:
I saw this, great idea.

flothesof, 2019-09-16:
Some more random things: the README mentions IMAS, but I had to google what it is. Suggestion: add a link to an official IMAS document (I found this published paper about it: https://iopscience.iop.org/article/10.1088/0029-5515/55/12/123006/meta but I guess you guys have better link ideas)

lasofivec, 2019-09-16:
For the record, I would say this is left to do:
- [ ] Add IMAS reference
- [ ] Add instructions on how to install
- [ ] Add examples that create images in README
- [ ] Add images/gallery corresponding to those images

Didou09, 2019-11-22:
@lasofivec , I guess with the last progress we can probably consider this issue as "fixed in devel" to include it in the next full release 1.4.2, do you agree ?

We'll open a new one for documentation updates for the following version (1.4.3), just to keep track of progress version-per-version

lasofivec, 2019-11-25:
ok !

closed 2020-04-07 by Didou09