ToFuProject/tofu #183: Mac OSX - OpenMP - reduction

lasofivec, 2019-09-11:
After much consideration, I am opening an issue here even though it is **not** a **tofu** error but a problem of configuration between OSX, Cython an OpenMP, which leads to **wrong** results (tests failing) even if there is no issue in **tofu**.

The first issue I found is that with ``homebrew``'s ``gcc-8``, OpenMP reductions are not being run in parallel but repeated. 
Some example can be found here:

https://stackoverflow.com/questions/54776301/cython-prange-is-repeating-not-parallelizing
https://github.com/bearnshaw/mult-cython
https://github.com/cython/cython/issues/2316

lasofivec, 2019-09-11:
The solution seems to be to use ``conda``'s ``gcc`` and make sure this is the one being used by default, ie. ``which gcc`` should give something among the lines:
``/Users/<username>/anaconda3/envs/<yourenv>/bin/gcc``.

Since **tofu** used to fix the ``gcc`` compiler only for the compatibility with OSX, the ``setup.py`` can be updated so that it doesn't anymore. I will add the specification for the ``gcc`` compiler to the documentation.
This will also simplify the installation on ITER as Simon Pinches has to do a patch currently for the installation.

Didou09, 2019-09-12:
Fixed, in devel, by PR #184

PR #184 (2019-09-12): update in setup.py
Fixes #183 

- Now we don't specify the `gcc` compiler for Mac OSX (or any OS). It will use the system's default compiler. This fixes the above mentioned Issue as well as prevent the need for a patch (on ITER servers)

- Passed **black** on the ``setup.py`` file.
closed 2020-04-07 by Didou09