ToFuProject/tofu #186: In _core.py: unused variable

lasofivec, 2019-09-11:
https://github.com/ToFuProject/tofu/blob/807da74401e4e96f849127e7dae8afcd558f9dc0/tofu/geom/_core.py#L424-L425

The variable here is declared but not used

lasofivec, 2019-09-11:
I can erase this variable (and thus also `Typesu` which is only needed for `lc` declaration.
@Didou09  can you confirm it? If so, I can do it

Didou09, 2019-09-11:
Not sure, I need to think about it, it might be that there is typo on the line below and that it should be saying something like :
``` assert np.all(np.any(Typesu, axis=0)) ```

In which case lc and Typesu are necessary,
Give me one day more

lasofivec, 2019-09-11:
Since `Typesu` is the same as `Types` but after `np.unique` seems to me that both statements are equivalent.
Meaning `assert np.all(np.any(Typesu, axis=0))` is equivalent to `assert np.all(np.any(Types, axis=0))`, and the second one should be more general.

closed 2020-04-07 by Didou09