ToFuProject/tofu #190: In _core.py: set of variables related issues

lasofivec, 2019-09-11:
In order not to spam everyone and create `n` issues, I pushed my branch `Issue30_pep8` parsed with a `flake8` git hook gives the following errors:

> tofu/geom/_core.py:2083:9: E743 ambiguous function definition 'I'
> tofu/geom/_core.py:2090:21: E741 ambiguous variable name 'I'
> tofu/geom/_core.py:2098:13: E741 ambiguous variable name 'I'
> tofu/geom/_core.py:2100:13: E741 ambiguous variable name 'I'
> tofu/geom/_core.py:3907:20: F821 undefined name 'self'
> tofu/geom/_core.py:3912:16: F821 undefined name 'self'
> tofu/geom/_core.py:3914:36: F821 undefined name 'self'
> tofu/geom/_core.py:4822:24: F821 undefined name 'Dsi'
> tofu/geom/_core.py:4822:29: F821 undefined name 'usi'
> tofu/geom/_core.py:5012:24: F821 undefined name 'force'
> tofu/geom/_core.py:5078:20: F821 undefined name 'Nonei'
> tofu/geom/_core.py:5464:17: F841 local variable 'kRMin' is assigned to but never used
> tofu/geom/_core.py:5858:17: F841 local variable 'kh' is assigned to but never used
> tofu/geom/_core.py:6203:34: F821 undefined name 'fkwdargs'

I verified all of them and they are non-trivial (except maybe from ``Nonei`` which is probably a typo).
@Didou09 I'll leave you the handling of the branch, so you can change it as you want.

lasofivec, 2019-11-04:
Again, these errors found by flake look like they are actually bugs, or could be bugs in the future:

> tofu/geom/_core.py:2019:9: E743 ambiguous function definition 'I'
> tofu/geom/_core.py:2026:21: E741 ambiguous variable name 'I'
> tofu/geom/_core.py:2034:13: E741 ambiguous variable name 'I'
> tofu/geom/_core.py:2036:13: E741 ambiguous variable name 'I'
> tofu/geom/_core.py:4695:24: F821 undefined name 'Dsi'
> tofu/geom/_core.py:4695:29: F821 undefined name 'usi'
> tofu/geom/_core.py:4885:24: F821 undefined name 'force'
> tofu/geom/_core.py:4948:20: F821 undefined name 'Nonei'
> tofu/geom/_core.py:5334:17: F841 local variable 'kRMin' is assigned to but never used
> tofu/geom/_core.py:5746:13: F841 local variable 'indt' is assigned to but never used
> tofu/geom/_core.py:5871:17: F841 local variable 'kh' is assigned to but never used
> tofu/geom/_core.py:6217:34: F821 undefined name 'fkwdargs'

Didou09, 2019-11-04:
Good point, will look at it tomorrow

Didou09, 2019-11-05:
@lasofivec since Pr #240 is now merged and closed, I label this issue "fixed in devel", ok for you ?

lasofivec, 2019-11-05:
No... I don't see the relationship between the two 😅 
This issue is related to the two comments above... which have nothing to do with the additional files that were introduced at another PR into devel.

lasofivec, 2019-11-05:
I just realized you meant it was solved in PR #236
Then yes :)

Didou09, 2019-11-05:
Yes, sorry, I wasn't clear:
PR #236 also solves this issue

closed 2020-04-07 by Didou09