ToFuProject/tofu #198: Create conda package for Mac and Windows

lasofivec, 2019-09-16:
**tofu** is only packaged for Linux-64 bits, so

``conda install -c tofuproject tofu``

only works on this platform. We could package it for `Mac OSX` and soon for `Windows` as well.

flothesof, 2019-09-23:
I guess this opens up the question for packaging: conda or pip wheels? or both? If you plan on doing windows support, I suppose we will also need to look for Windows CI (using CircleCI?).

Another thought on the topic would be: we could be using the conda-forge ecosystem to provide builds for tofu. See https://conda-forge.org/#about

lasofivec, 2019-09-24:
Should we chat about this on [gitter](https://gitter.im/ToFuProject/community)? or by videoconference?

I think we should discuss all the available options with @Didou09 , would you like to be a part of the meeting @flothesof ? The more the merrier :)

Didou09, 2019-09-27:
Great idea to use the conda-forge tools / environment !
And also to do pip wheels, the more options for installation the better, feedback shows that each user has his own constraints, tofu should be easy to install for everyone, so the more options available the better.

Didou09, 2019-11-20:
All conda / travis issues now referenced in Issue #259

closed 2019-11-20 by Didou09