ToFuProject/tofu #200: source files missing in new-doc branch

Didou09, 2019-09-27:
I pulled the new-doc branch locally to work on it and update some installation instructions for ITER users (to recommend using conda environments instead of modules which are not operational yet).

But in doc/source/, the installtion_tabs_source file only exists in html format, any chance to get the corresponding rst source or do we have to edit the html directly ?

lasofivec, 2019-09-27:
This is a automatically generated files. How did you build the doc?

Didou09, 2019-09-27:
I haven't built it yet, just pulled the branch, seems like these html were brought along by the pull, sithout the corresponding source.

This is consistent with the content of doc/source on Github : the html is there but there is no source.

lasofivec, 2019-09-27:
Oh sorry. I thought it was another file (I'm at the airport waiting for my flight)
Yes you have to modify the html directly. It was the cleanest best way to do it.

flothesof, 2019-09-28:
Hi all,
I hadn't actually seen the source for the installation files. After inspecting it, my opinion is the following: although the tabbed installation instructions look nice, I would advise against using raw HTML like @lasofivec did. 
The main reason is that it's not easy to edit (lots of HTML tags everywhere, you have to know one more building block), compared to a .rst file with some sections (I'm quite sure we can have a mini table-of-contents with clickable anchor links inside a .rst file instead of pure HTML).

lasofivec, 2019-10-22:
Seems to me we can talk about this on the [gitter](https://gitter.im/ToFuProject/community) ?
In any case this issue was never really a issue so i'm closing it.

closed 2019-10-22 by lasofivec