ToFuProject/tofu #201: tofu 1.4.1 installation on Eurofusion Gateway failing

Didou09, 2019-09-27:
The pip installation on the gateway failed.

In fact, it succeeded, but there is an error at import time, suggesting something went wrong at compilation time.
The error message is somewhat similar to the first one obtained on ITER:

```
In [1]: import tofu
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
/gw/switm/python/3.6.0.b3/lib/python3.6/site-packages/tofu/geom/_core.py in <module>()
     29 try:
---> 30     import tofu.geom._def as _def
     31     import tofu.geom._GG as _GG

AttributeError: module 'tofu' has no attribute 'geom'

During handling of the above exception, another exception occurred:

TypeError                                 Traceback (most recent call last)
/gw/switm/python/3.6.0.b3/lib/python3.6/site-packages/tofu/geom/_GG.py in <module>()
     14     try:
---> 15         from tofu.geom._GG03 import *
     16     except Exception:

/gw/switm/python/3.6.0.b3/lib/python3.6/site-packages/tofu/geom/_GG03.pyx in init tofu.geom._GG03 (tofu/geom/_GG03.cpp:103443)()
----> 1 # cython: boundscheck=False
      2 # cython: wraparound=False
      3 # cython: initializedcheck=False

TypeError: C function tofu.geom._basic_geom_tools.compute_hypot has wrong signature (expected arrayobject *(__Pyx_memviewslice const , __Pyx_memviewslice const , struct __pyx_opt_args_4tofu_4geom_17_basic_geom_tools_compute_hypot *__pyx_optional_args), got arrayobject *(__Pyx_memviewslice, __Pyx_memviewslice, struct __pyx_opt_args_4tofu_4geom_17_basic_geom_tools_compute_hypot *__pyx_optional_args))

During handling of the above exception, another exception occurred:

ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-1-fb62ffe09d58> in <module>()
----> 1 import tofu

/gw/switm/python/3.6.0.b3/lib/python3.6/site-packages/tofu/__init__.py in <module>()
     70 from tofu.utils import save, load, load_from_imas, calc_from_imas
     71 import tofu._plot as _plot
---> 72 import tofu.geom as geom
     73 import tofu.data as data
     74

/gw/switm/python/3.6.0.b3/lib/python3.6/site-packages/tofu/geom/__init__.py in <module>()
     10 """
     11
---> 12 from tofu.geom._core import PlasmaDomain, Ves, PFC, CoilPF, CoilCS, Config
     13 from tofu.geom._core import Rays, CamLOS1D, CamLOS2D
     14 from . import utils

/gw/switm/python/3.6.0.b3/lib/python3.6/site-packages/tofu/geom/_core.py in <module>()
     34 except Exception:
     35     from . import _def as _def
---> 36     from . import _GG as _GG
     37     from . import _comp as _comp
     38     from . import _plot as _plot

/gw/switm/python/3.6.0.b3/lib/python3.6/site-packages/tofu/geom/_GG.py in <module>()
     15         from tofu.geom._GG03 import *
     16     except Exception:
---> 17         from _GG03 import *

ModuleNotFoundError: No module named '_GG03'
```

olivhoenen, 2019-09-27:
@Didou09 for this build GCC which is being used is in version 6.1.0 and cython in version 0.29.13

olivhoenen, 2019-09-27:
I've been checking further this issue, when tried to reinstall with -v and --force-reinstall I saw some errors messages with scipy that was required in a different version for another package. After removing this package and installing tofu it seems to work better.

olivhoenen, 2019-09-27:
I've been successfully running the first part of the 5 min tutorial, so I guess basic install is ok:

```
In [4]: import tofu as tf
No module named 'pywed'
No module named 'pywed'
/gw/switm/python/3.6.0.b3/lib/python3.6/site-packages/tofu/mag/__init__.py:28: UserWarning: Traceback (most recent call last):
  File "/gw/switm/python/3.6.0.b3/lib/python3.6/site-packages/tofu/mag/magFieldLines.py", line 30, in <module>
    from tofu.mag.mag_ripple.mag_ripple import mag_ripple
ModuleNotFoundError: No module named 'tofu.mag.mag_ripple.mag_ripple'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/gw/switm/python/3.6.0.b3/lib/python3.6/site-packages/tofu/mag/__init__.py", line 13, in <module>
    from tofu.mag.magFieldLines import *
  File "/gw/switm/python/3.6.0.b3/lib/python3.6/site-packages/tofu/mag/magFieldLines.py", line 32, in <module>
    from mag.mag_ripple.mag_ripple import mag_ripple
ModuleNotFoundError: No module named 'mag'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/gw/switm/python/3.6.0.b3/lib/python3.6/site-packages/tofu/mag/magFieldLines.py", line 30, in <module>
    from tofu.mag.mag_ripple.mag_ripple import mag_ripple
ModuleNotFoundError: No module named 'tofu.mag.mag_ripple.mag_ripple'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/gw/switm/python/3.6.0.b3/lib/python3.6/site-packages/tofu/mag/__init__.py", line 15, in <module>
    from .magFieldLines import *
  File "/gw/switm/python/3.6.0.b3/lib/python3.6/site-packages/tofu/mag/magFieldLines.py", line 32, in <module>
    from mag.mag_ripple.mag_ripple import mag_ripple
ModuleNotFoundError: No module named 'mag'

    => the optional sub-package tofu.mag is not usable

  warnings.warn(msg)

In [5]: configB2 = tf.geom.utils.create_config("B2")

In [6]: configB2.plot()
Out[6]:
[<matplotlib.axes._axes.Axes at 0x2ba041e82ef0>,
 <matplotlib.axes._axes.Axes at 0x2ba0422ec438>]
```

Didou09, 2019-09-27:
Ok, so it suggests it was dependencies-related.
We still need to investigate because installing tofu should be easy despite its co-existence with other packages.

**Regarding the tutorial:**

tofu has 2 optional sub-packages : tofu.imas2tofu and tofu.mag

They are optional in the sense that they rely on specific libraries (imas and a fortran library) that not everyone has.
When importing tofu, if one of these dependencies is detected as absent, tofu throws a warning specifying that the corresponding optional sub-package is not usable, but does not throw an error.
This way the user is informed of why the sub-package cannot be used, but he can still use the rest of tofu normally. 
The warning includes the traceback to make debugging easier in case the user wants to find out what happened.

Thanks very much for your detailed feedback and help @olivhoenen, we can consider basic install is fine until @lasofivec comes back and has a closer look at the first failed attempt (at least to make installing future releases more robust).

lasofivec, 2019-10-24:
> I've been checking further this issue, when tried to reinstall with -v and --force-reinstall I saw some errors messages with scipy that was required in a different version for another package. After removing this package and installing tofu it seems to work better.

Thanks for the feedback @olivhoenen.
So, in conclusion the installation (or update?) from pip didn't work but forcing the reinstallation/compilation of the library made it work?
Do you remember which package(s) had issues as you stated above? 
It wasn't clear to me if it were tofu's dependencies, tofu modules or un related installed packages.

olivhoenen, 2019-10-24:
Hi @lasofivec, yes the aforementioned package was EasyVVUQ (which is still in alpha stage, and I'm not 100% sure which version I had back then) developed in another project and which has this [this](https://github.com/UCL-CCS/EasyVVUQ/blob/dev/requirements.txt) of requirements. Not that I am not so sure if removing this package (and it's scipy related conflict) was the only reason for this issue, as --force-reinstall also updated quite a few other packages.

lasofivec, 2019-10-24:
I'm thinking it is more the `--force-reinstall` that did the trick.
On another platform installing by using the `setup tools` a `clean` was necessary before the installation was successful. I am not sure why it didn't work out of the box though.

closed 2019-11-01 by lasofivec