ToFuProject/tofu #211: Tools for CamLOS closest points

Didou09, 2019-10-08:
Create two methods with easy / explicit names in Rays:
* one to compute the geometrically closest point of each LOS to an axis (using self.set_dsino() and returning only the distance, theta and pts
* one to compute the points of minimum rho from a Plasma2D instance and returning rho, theta and the pts

Didou09, 2019-10-08:
Fixed in devel by PR #212

lasofivec, 2019-10-23:
> Create two methods with easy / explicit names in Rays:
> 
>     * one to compute the geometrically closest point of each LOS to an axis (using self.set_dsino() and returning only the distance, theta and pts

What do you mean by axis?
In any case if an axis is a line, then there is already a function that is similar to this probably.

Didou09, 2019-10-24:
Yes, the functions in _GG are all already existing and operational, I just added extra methods to the Rays class with more explicit names from a tokamak point of view (axis = magnetic axis = circle in 3d)

Fixed

closed 2020-04-07 by Didou09