ToFuProject/tofu #217: Plotting signal with t=None

lasofivec, 2019-10-22:
*See PR #207 for more details*

When plotting a signal computed with `LOS_calc_signal(..., t=None, ...)`, the plot fails with:

> 
> Traceback (most recent call last):
>   File "examples/tutorials/plot_custom_emissivity.py", line 80, in <module>
>     sig.plot()
>   File "/home/mendoza/miniconda3/envs/tofu3/lib/python3.7/site-packages/tofu-.1.4.1_145_ge5ffef4-py3.7-linux-x86_64.egg/tofu/data/_core.py", line 1657, in plot
>     draw=draw, connect=connect)
>   File "/home/mendoza/miniconda3/envs/tofu3/lib/python3.7/site-packages/tofu-.1.4.1_145_ge5ffef4-py3.7-linux-x86_64.egg/tofu/data/_plot.py", line 161, in Data_plot
>     fontsize=fontsize, draw=draw, connect=connect)
>   File "/home/mendoza/miniconda3/envs/tofu3/lib/python3.7/site-packages/tofu-.1.4.1_145_ge5ffef4-py3.7-linux-x86_64.egg/tofu/data/_plot.py", line 879, in _DataCam12D_plot
>     groupinit='time', follow=True)
>   File "/home/mendoza/miniconda3/envs/tofu3/lib/python3.7/site-packages/tofu-.1.4.1_145_ge5ffef4-py3.7-linux-x86_64.egg/tofu/utils.py", line 2681, in __init__
>     out = self._checkformat_dgrouprefaxobj(dgroup, dref, dobj, dax, ddata)
>   File "/home/mendoza/miniconda3/envs/tofu3/lib/python3.7/site-packages/tofu-.1.4.1_145_ge5ffef4-py3.7-linux-x86_64.egg/tofu/utils.py", line 3034, in _checkformat_dgrouprefaxobj
>     fgetval = get_valf(val, lrefs, linds)
>   File "/home/mendoza/miniconda3/envs/tofu3/lib/python3.7/site-packages/tofu-.1.4.1_145_ge5ffef4-py3.7-linux-x86_64.egg/tofu/utils.py", line 2468, in get_valf
>     assert ndim >= len(lrids)
> AssertionError

**Note:** there was already another bug in `devel` that I corrected in `bf_signal_tnone`. Please fix this issue in the same branch.

Didou09, 2019-11-06:
The problem was mostly due to:
* np.squeeze being used on a time of size = 1 
* a data array of shape (1, n) which was being flattened by np.squeeze
* np.digitize which was trying to get digitize a time vector of size = 1

=> deactivated the call to np.squeeze, test on size of t introduced

lasofivec, 2019-11-06:
so we can close this issue ?

closed 2020-04-07 by Didou09