ToFuProject/tofu #219: Cannot open saved camera object

munechika-koyo, 2019-10-24:
A CamLOS2D object was saved and the name was changed 👍 

```
# make camera object
cam = tfg.utils.create_CamLOS2D(case='C1', dcam=dcam, Exp=config.Id.Exp, Name='C1', Diag='test', config=config)

# save camera
cam.save(name='ITER_test_camera_config',path='output')
```

And now it cannot be opened any more, the following error shows up:

```
In [2]: cam = tf.load('public/MyTofu/output/ITER_test_camera_config.npz')
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-2-058b366e5346> in <module>
----> 1 cam = tf.load('public/MyTofu/output/ITER_test_camera_config.npz')

~/.conda/envs/tofu2/lib/python3.6/site-packages/tofu/utils.py in load(name, path, strip, verb)
    475
    476         # Recreate from dict
--> 477         mod = importlib.import_module( 'tofu.%s'%dd['dId_dall_Mod'] )
    478         cls = getattr( mod, dd['dId_dall_Cls'] )
    479         obj = cls(fromdict=dd)

~/.conda/envs/tofu2/lib/python3.6/importlib/__init__.py in import_module(name, package)
    124                 break
    125             level += 1
--> 126     return _bootstrap._gcd_import(name[level:], package, level)
    127
    128

~/.conda/envs/tofu2/lib/python3.6/importlib/_bootstrap.py in _gcd_import(name, package, level)

~/.conda/envs/tofu2/lib/python3.6/importlib/_bootstrap.py in _find_and_load(name, import_)

~/.conda/envs/tofu2/lib/python3.6/importlib/_bootstrap.py in _find_and_load_unlocked(name, import_)

ModuleNotFoundError: No module named 'tofu.None'
```

Any way to fix this ?  (tofu 1.4.1)

Didou09, 2019-11-04:
Hello @munechika-koyo , I am trying to debug this issue
I need to reproduce the error on my session
But I don't have access to your saved file:

/home/ITER/munechk/public/MyTofu/output/ITER_test_camera_config.npz

Could you make sure that your /public/ repository is accessible to others ?
Ithink the command is something like:

chmod -R ugo+rx /home/ITER/munechk/public/

see:
https://www.linux.com/tutorials/how-manage-file-and-folder-permissions-linux/

munechika-koyo, 2019-11-04:
I gave the accessibility to the folder.
Please confirm it.

munechika-koyo, 2019-11-04:
> Hello @munechika-koyo , I am trying to debug this issue
> I need to reproduce the error on my session
> But I don't have access to your saved file: [...]

munechika-koyo, 2019-11-04:
I am sorry to accidentally push the close button.

Didou09, 2019-11-04:
Does not seem to work yet

munechika-koyo, 2019-11-04:
I realized that my folder "munechk" wasn't given such permission.
I gave it.

Didou09, 2019-11-04:
Ok thanks, it's working

Turns out I can't reproduce the error, when I try to lad the camera with tofu, it works fine:

```
[1] import tofu as tf
[2] cam = tf.load('/home/ITER/munechk/public/MyTofu/output/ITER_test_camera_config.npz')
Loaded from:
    /home/ITER/munechk/public/MyTofu/output/ITER_test_camera_config.npz
[3]
```

Did you change something to this file since we last tried together ?
Can you try again from a fresh console ?

munechika-koyo, 2019-11-04:
yes! It works well also on my PC when I tried again.
But, I didn't change it at all.
I wonder what happened to tofu since then...

Didou09, 2019-11-04:
I'm not sure really
I'm closing this issue since it looks like the problem does not really exist, and we'll re-open it if it happens again (if it happens again, try to give as many details as possible so we can reproduce it)

closed 2019-11-04 by Didou09