ToFuProject/tofu #220: Integrate ITER new config

munechika-koyo, 2019-10-24:
I have extracted a more accurate configuration for ITER using CAD data, as can be seen below:

It would be nice to make it available in tofu for other users

lasofivec, 2019-10-28:
Dear @munechika-koyo,
That looks amazing. It looks like you are ready to do your first pull-request !
Did you created your own branch or did you fork the project?
I am going to add the instructions on how to create a pull request :)

munechika-koyo, 2019-10-29:
> Dear @munechika-koyo,
> That looks amazing. It looks like you are ready to do your first pull-request !
> Did you created your own branch or did you fork the project?
> I am going to add the instructions on how to create a pull request :)

Yes! I already create my own branch named "Issue220_NewITERConfig".
Also, Didier told me how to create a pull request, so that would be fine :)

Didou09, 2019-10-29:
Thanks Koyo for this nice addition to tofu,
Laura, if you can still add some instructions on for doing the pull request I feel it is still a very good ideas, both as a reminder of good practices and for newcomers.
In particular to make sure contributors do the pull request to devel and not to master directly, at least.
Many thanks !

lasofivec, 2019-11-01:
Done, documentation is up to date.

closed 2020-04-07 by Didou09