ToFuProject/tofu #224: Get rid of the setup.py's bag

munechika-koyo, 2019-10-30:
When I install the tofu into Windows PC using setup.py, it doesn't work well and I found the bags.
So, I try to get rid of them in the setup.py file.

munechika-koyo, 2019-10-30:
It seems that they have something to do with getting the path and typo.

lasofivec, 2019-10-31:
Could you please copy paste the error ?
As well as the command that you are using to compile the library (eg. ``python setup.py install`` or ``python setup.py build_ext --inplace``) ?

munechika-koyo, 2019-11-03:
I tried both of them, and the same error message appeared:
```
$ python setup.py install
cython version = 0.29.12
numpy  version = 1.16.4
cython version = C:\Users\Koyo\Anaconda3\lib\site-packages\Cython\__init__.py
numpy  version = C:\Users\Koyo\Anaconda3\lib\site-packages\numpy\__init__.py
................ checking if openmp installed...
................ checking if openmp installed... >  False

Version for setup.py :  1.4.0-745-ge908e77

Traceback (most recent call last):
  File "setup.py", line 221, in <module>
    _README = [ff for ff in os.listdir(os.path.abspath(__file__))
NotADirectoryError: [WinError 267] ▒f▒B▒▒▒N▒g▒▒▒▒▒▒▒▒▒▒▒ł▒▒B: 'D:\\Documents\\tofu\\setup.py'
```
sorry that the text is garbled because that was caused by Japanese font.
But, it means that " the following path does not exist".

I think it is caused by using `os.listdir(os.path.abspath(__file__)`. This seems to make something wrong in windows.
I personally recommend using `os.get_cwd`

flothesof, 2019-11-04:
Hi there!
I'm surprised by @munechika-koyo 's error.
According to the docs, the recommended way of installing is pip install -e .[dev] 
Does that work better on your system? 
This works without a problem on my machine in fact. 
Regards 
Florian

Didou09, 2019-11-04:
On the one hand, @flothesof is right, for windows users, pip is the recommended way to go.
On the other hand, @munechika-koyo, as a contributor, allows us to test tofu installation in various configurations, and the 'python setup.py' should work, in principle...

We should try to make sure it works, installing tofu should be as easy as possible no matter how it is installed (conda, pip, python setup.py...)

However, @munechika-koyo , I cannot agree on your coding suggestion because both commands do not give the same result:

Here we want get the directory of the **file** (i.e. in which the module being executed in located)
`os.path.abspath(__file__)`

Here we get the current working directory of the **user** (i.e.: the directory where the user is, not where the file is)
`os.getcwd()`

You can check this by creating, on your home path, a temporeray directory sandbox/ and inside create a temporary python module __init__.py (to make it a package) and a sandbox.py.
In sandbox.py, write

```
import os
print('os.getcwd()              :', os.getcwd())
print('os.path.abspath(__file__):', os.path.abspath(__file__))
print('os.path.dirname(__file__):', os.path.dirname(__file__))
```

Then go back to your home (i.e.: get out of sandbox directory), and open an python console, and import the sandbox module:

import sandbox.sandbox

You'll see the results are not the same, on my (CEA) session it gives:
```
os.getcwd()              :  /Home/DV226270
os.path.abspath(__file__):  /Home/DV226270/sandbox/sandbox.py
os.path.dirname(__file__):  sandbox
```

munechika-koyo, 2019-11-05:
I'm sorry, I misunderstand that. I totally agree with you.
And I didn't notice that 221st line in the setup.py in the Issue224_setup branch :

`_README = [ff for ff in os.listdir(os.path.abspath(__file__))`,

is fixed to 

`_README` = [ff for ff in `os.listdir(os.path.abspath(os.path.dirname(__file__)))`

in the master branch.

This works well now.

Although it is detailed, `os.path.abspath(os.path.dirname(__file__))` is defined as `_Here`  in preveious
line, so it might be better to replace.

lasofivec, 2019-11-05:
Yes ! I agree with Koyo :)

So this issue was fixed in PR #

closed 2019-11-05 by lasofivec