ToFuProject/tofu #227: Wrong naming of PFC files for B4 configuration

lasofivec, 2019-11-03:
I tried to construct the new ITER configuration "B4" by doing the following:

```
In [2]: configB4 = tf.geom.utils.create_config("B4")
```

and got 

``` 
---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
<ipython-input-2-39b2cf68bbcd> in <module>
----> 1 configB4 = tf.geom.utils.create_config("B4")

~/miniconda3/envs/tofu3/lib/python3.7/site-packages/tofu-.1.4.1_145_ge5ffef4-py3.7-linux-x86_64.egg/tofu/geom/utils.py in create_config(case, Exp, Type, Lim, Bump_posextent, R, r, elong, Dshape, divlow, divup, nP, out, SavePath, path)
    758 
    759     if case is not None:
--> 760         conf = _create_config_testcase(config=case, out=out, path=path)
    761     else:
    762         poly, pbump, pbaffle = _compute_VesPoly(R=R, r=r, elong=elong, Dshape=Dshape,

~/miniconda3/envs/tofu3/lib/python3.7/site-packages/tofu-.1.4.1_145_ge5ffef4-py3.7-linux-x86_64.egg/tofu/geom/utils.py in _create_config_testcase(config, out, path, dconfig)
    685                 msg += "    Criteria: [%s, %s]\n"%(cc,ss)
    686                 msg += "    Matching: "+"\n              ".join(ff)
--> 687                 raise Exception(msg)
    688             pfe = os.path.join(path,ff[0])
    689             obj = eval('_core.'+cc).from_txt(pfe, Name=ss, Type='Tor',

Exception: No / several matching files
  Folder: /home/lasofivec/miniconda3/envs/tofu3/lib/python3.7/site-packages/tofu-.1.4.1_145_ge5ffef4-py3.7-linux-x86_64.egg/tofu/geom/inputs
    Criteria: [PFC, BLK1]
    Matching: TFG_PFC_ExpITER_BLK16.txt
              TFG_PFC_ExpITER_BLK15.txt
              TFG_PFC_ExpITER_BLK14.txt
              TFG_PFC_ExpITER_BLK18.txt
              TFG_PFC_ExpITER_BLK13.txt
              TFG_PFC_ExpITER_BLK12.txt
              TFG_PFC_ExpITER_BLK11.txt
              TFG_PFC_ExpITER_BLK1.txt
              TFG_PFC_ExpITER_BLK10.txt
              TFG_PFC_ExpITER_BLK17.txt
```
                                                                                                                                            
The reason for this is error is that when loading the files it will match any file starting with ``PFC`` and containing ``BLK1`` in the name of the file. So it matches many additional files.
This is easily solved by renaming the file

> TFG_PFC_ExpITER_BLK1.txt

to 

> TFG_PFC_ExpITER_BLK**0**1.txt

as well as all names of `BLK` files with number $<10$ and respectively update in `tofu/geom/utils.py` the name of the elements.

munechika-koyo, 2019-11-04:
yes, that's right. I noticed the error also, and I already changed them tp this commit:
https://github.com/ToFuProject/tofu/commit/5df25148ee639767cc0bf0a51aa42917f9e8738c

could you make sure if this would be fine?

lasofivec, 2019-11-04:
It's almost there :) 
I added a couple of comments on each commit. Could you take a look please ?

PS: could you also do a pull request of this branch into devel ?

closed 2020-04-07 by Didou09