ToFuProject/tofu #230: Stop Python 2.7 support

Didou09, 2019-11-04:
From the next version, tofu will not support python 2.7 anymore
This means we can remove the python 2.7 builds in travis.yml

lasofivec, 2019-11-04:
There's already a branch with this in mind see SHA: 7e321f6c3d49303ecd05cbbc8ef471934e970c81

Didou09, 2019-11-04:
Ok, but please next time open a dedicated issue and name the branch accordingly, it easier to follow.
Do you want to consider this issue as being associated to the branch you mention then ?

lasofivec, 2019-11-04:
done

Didou09, 2019-11-04:
Just to preserve the workflow:
* issue labelled "fixed in devel" when the PR accepted
* issue closed only when solved in the last release or when it was opened by mistake (e.g.: the bug does not really exist...)

This way we'll easily get a list of all additions to a release simply by collecting all issues labelled "fixed in devel", and we'll close them all after the release is done

=> I re-open the issue and label it "fixed in devel", to keep track as a good practice

closed 2020-04-07 by Didou09