ToFuProject/tofu #235: magfieldline should plot config with reversed x axis

Didou09, 2019-11-04:
Plotting the config, as seen from the interior of the vessel, vs increasing toroidal angle is un-intuitive because it reverses the left-right perception (if the user imagines himself inside the vessel.

Such plots should have a reversed x-axis

Didou09, 2019-11-06:
As shown below, the perception issue is fixed in PR #226 by commit 
https://github.com/ToFuProject/tofu/pull/226/commits/de5dc6bb7d3578dd9da36c23d469e9dba03058c3

But the pitch angle issue remains: @jmoralesFusion do you think the pitch angle should be opposite (i.e. field lines should go up from positive phi towards negative phi) ?
Could you double-check the field-line tracing routine and signs conventions ?

[image]

Guillaume-Urban, 2019-11-07:
Bravo, ces modifs sont impeccables. Reste peut être un détail qui pourrait être utile, ce serait d'avoir quelque part accès facilement aux coordonnées R,Z du centre de chaque monobloc ainsi que des strike points, ce qui permet de mieux choisir la ligne que l'on souhaite tracer pour aider l'analyse par exemple des données de sonde de Langmuir ou THR ... Choisir à tâtons peut s'avérer complexe tant qu'il n'y a pas de calcul d'intersection.
[image]

Didou09, 2019-11-07:
Hi @Guillaume-Urban ,
Thanks, 
So several points to answer your question:

- The boundary used to compute field-line limits is now approximate, and in particular it does not take into account some elements like the divertor. In future version, a more detailed boundary including the divertor and other elements will be implemented, when we can find some time for it (maybe by the end of 2019, hopefully).

- If you use tofuplot (i.e. bash command) you don't have access to a lot of details / tools. Th ebash commands are just shortcuts to quickly get a graph. However, if you use tofu from an ipython console, you get much more tools. In particular you can load the Config (i.e. a class in tofu representing the geometrical configuration of the vacuum vessel nd its components) and plot the element you want, with the indices of each of its segments:

Using tofu >= 1.4.1, you can first load the configuration of WEST with the required level of details ('B2', but the name will be more explicit in future version), and get an overview of what it contains (components are separated by class: Ves or PFC and name).

```
In [1]: import tofu as tf
In [2]: conf = tf.geom.utils.create_config('B2')

In [3]: conf
Out[3]: 
tot. Struct  tot. occur  tot. points
-----------  ----------  -----------
9            17          690        

class  Name           SaveName                                                    nP   noccur  mobile  color                 visible
-----  -------------  ----------------------------------------------------------  ---  ------  ------  --------------------  -------
Ves    V2             TFG_Ves_ExpWEST_V2_sh00000_Vers1.4.1-4-g7cfbe9f             59   0       False   (0.0, 0.0, 0.0, 1.0)  True   
PFC    BaffleV1       TFG_PFC_ExpWEST_BaffleV1_sh00000_Vers1.4.1-4-g7cfbe9f       31   0       False   (0.8, 0.8, 0.8, 0.8)  True   
PFC    DivUpV2        TFG_PFC_ExpWEST_DivUpV2_sh00000_Vers1.4.1-4-g7cfbe9f        8    0       False   (0.8, 0.8, 0.8, 0.8)  True   
PFC    DivLowITERV2   TFG_PFC_ExpWEST_DivLowITERV2_sh00000_Vers1.4.1-4-g7cfbe9f   149  0       False   (0.8, 0.8, 0.8, 0.8)  True   
PFC    BumperInnerV1  TFG_PFC_ExpWEST_BumperInnerV1_sh00000_Vers1.4.1-4-g7cfbe9f  33   6       False   (0.8, 0.8, 0.8, 0.8)  True   
PFC    BumperOuterV1  TFG_PFC_ExpWEST_BumperOuterV1_sh00000_Vers1.4.1-4-g7cfbe9f  32   1       False   (0.8, 0.8, 0.8, 0.8)  True   
PFC    IC1V1          TFG_PFC_ExpWEST_IC1V1_sh00000_Vers1.4.1-4-g7cfbe9f          126  2       False   (0.8, 0.8, 0.8, 0.8)  True   
PFC    IC2V1          TFG_PFC_ExpWEST_IC2V1_sh00000_Vers1.4.1-4-g7cfbe9f          126  2       False   (0.8, 0.8, 0.8, 0.8)  True   
PFC    IC3V1          TFG_PFC_ExpWEST_IC3V1_sh00000_Vers1.4.1-4-g7cfbe9f          126  2       False   (0.8, 0.8, 0.8, 0.8)  True   

```

 Then you can plot it with indices=True to get the indiex of each segment (element='P' indicates you just want to plot the polygon), it returns a list of 2 axes:
`In [13]: lax = conf.PFC.DivLowITERV2.plot(indices=True, element='P')`

And get the index of the upper face of the monobloc of interest (e.g.: 131).
The center is the average of the 2 points constituting the segment of the polygon. 
You can then plot it on top of the previous figure

```
In [4]: r, z = np.mean(conf.PFC.DivLowITERV2.Poly_closed[:, 131:133], axis=1)
In [5]: lax[0].plot(r, z, '*r')
```

[image]

Didou09, 2019-11-20:
The original issue (plotting with reversed x axis) was fixed in devel by PR #226

closed 2020-04-07 by Didou09