ToFuProject/tofu #247: Bug in _GG_LOS_calc_signal()

Didou09, 2019-11-06:
Trying to debug PR #207, I realized the output of the script:
examples/tutorials/plot_custom_emissivity.py

Depends a lot on whether newcalc=True or False

True: (new optimized routines) - wrong result

False: (old reference routine) - good result

Didou09, 2019-11-06:
High priority, we shoul probably issue a new released when fixed, as current users may run into false results

Didou09, 2019-11-06:
Once the bug is identified, solved and similar routines are double-checked, I suggested adding a new unit test based on this particular tutorial to check that the results of newcalc=True and newcalc=False are very close / identical (use np.allclose()), what do you think ?

Didou09, 2019-11-17:
Fixed in devel by PR #249

jdhare, 2019-11-25:
Resolved in 1.4.2, many thanks!

Didou09, 2019-11-25:
Yes :-)
tofu 1.4.2 is currently in pre-release (alpha), we're testing its portability (ease of installation on several platfoms : Linux, Mac OS, Windows via pip, conda and easyconfig).

So a final (non-alpha) release 1.4.2 will come soon, but most of the bugs and new features are already available in the alpha, so don't hesitate to use it already (and to report any new problem by openeing an issue here). Thanks for your support and feeback @jdhare !

closed 2020-04-07 by Didou09