ToFuProject/tofu #248: assert ndim in [2,3] error in _checkformat_inputs_ddataRef

jdhare, 2019-11-08:
Hello,

I have generated our cameras for ITER, and I am trying to calculate the signal using calc_signal()

Even when using a dummy synthetic emission function such as: 
`def ff(Pts, t=None):
    return np.zeros(Pts.shape[1])+1
`
I still get the following error:

`c.calc_signal(ff, t=None)
Traceback (most recent call last):
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/IPython/core/interactiveshell.py", line 3326, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-110-851451bbde5c>", line 1, in <module>
    c.calc_signal(ff, t=None)
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/geom/_core.py", line 4227, in calc_signal
    osig = tfd.DataCam1D(**kwdargs)
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/data/_core.py", line 160, in __init__
    super(DataAbstract,self).__init__(**kwdargs)
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/utils.py", line 719, in __init__
    self._init(**kwdargs)
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/data/_core.py", line 610, in _init
    self._set_ddataRef(**kwddataRef)
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/data/_core.py", line 631, in _set_ddataRef
    lout = self._checkformat_inputs_ddataRef(**kwdargs)
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/data/_core.py", line 276, in _checkformat_inputs_ddataRef
    assert ndim in [2,3]
AssertionError`

The camera seems otherwise fine, I can plot its LOS on the vessel etc. Any ideas?

Didou09, 2019-11-08:
Yes, the emissivity function should always return a 2d array, where the first dimension is for time evolution.
Try redefining it as 
`def ff(Pts, t=None): return np.zeros((1, Pts.shape[1]))+1 `

It should solve your problem.

A caveat though: we very recently identified a bug in some the signal computation routines (see Issue #247 for more details).
We are on it and will release a corrective update in the days to come.
In the meantime, be sure to use the option newcalc=False as this will perform the signal computation with a different algorithm that is bug-free.

jdhare, 2019-11-11:
Hi Didier,
 
Now I'm getting:

`Traceback (most recent call last):
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/IPython/core/interactiveshell.py", line 3326, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-114-851451bbde5c>", line 1, in <module>
    c.calc_signal(ff, t=None)
  File "/home/jhare/.conda/envs/tofu/lib/python3.6/site-packages/tofu/geom/_core.py", line 4197, in calc_signal
    t=t, Ani=ani, fkwdargs=fkwdargs, Test=True)
  File "tofu/geom/_GG03.pyx", line 4414, in tofu.geom._GG03.LOS_calc_signal
  File "tofu/geom/_GG03.pyx", line 4324, in tofu.geom._GG03.check_ff
AssertionError: When fed a (3,N) np.array only, or if t is a scalar, ff must return a (N,) np.ndarray!
`

newcalc does not seem to be in the call signature for calc_signal(), where should I set it?

Didou09, 2019-11-11:
Ok sorry, can you specify the version of tofu you are using ?

```
import tofu as tf
print(tf.__version__)
```

jdhare, 2019-11-11:
Apologies, I was accidentally using 1.3. I've now switched to a newer version:
`import tofu as tf
print(tf.__version__)
1.4.0-444-g0fab6c4`

Now the error reads:
`
def ff(PtsXYZ, t=None):
    return np.zeros((1, PtsXYZ.shape[1]))+1.0

c.calc_signal(ff, t=None)
Traceback (most recent call last):
  File "/home/jhare/.conda/envs/tofu-dev/lib/python3.7/site-packages/IPython/core/interactiveshell.py", line 3326, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-16-851451bbde5c>", line 1, in <module>
    c.calc_signal(ff, t=None)
  File "/home/jhare/.conda/envs/tofu-dev/lib/python3.7/site-packages/tofu-1.4.0_444_g0fab6c4-py3.7-linux-x86_64.egg/tofu/geom/_core.py", line 4610, in calc_signal
    t=t, ani=ani, fkwdargs=fkwdargs, Test=True)
  File "tofu/geom/_GG03.pyx", line 2979, in tofu.geom._GG03.LOS_calc_signal
TypeError: must be real number, not NoneType`

Didou09, 2019-11-14:
Hi @jdhare , sorry for the delay, emergencies on diagnostics,
How did you install the new version ? conda ? pip ? git pull ? python setup.py install ?

There an issue being debugged currently that might be linked to this.
We'll release a new version very soon (in a matter of days), we'll make sure it also addresses your issue

lasofivec, 2019-11-15:
1.4.0 is not the latest released version of tofu, it's 1.4.1. But as Didier mentioned, we found (and solved) some bugs. We are bout to release a new version on tofu (1.4.2), and I just tried your example with that version of tofu, so it will work for you.

[image]
[image]

See https://tofuproject.github.io/tofu/auto_examples/tutorials/plot_custom_emissivity.html

jdhare, 2019-11-19:
Hi both, apologies for not using the latest version. I have upgraded now, but a quick scan of the open issues on github makes it look like I should wait for 1.4.2? Is that what you recommend?

closed 2020-04-07 by Didou09