ToFuProject/tofu #250: pip install dependencies missing

Didou09, 2019-11-15:
@rragona tried to install tofu on a Macbook using a fresh (virgin) conda virtual environment
* The conda package for Mac doesn't exist so he had to use a pip install (using the pip installed via conda)
* pip did not check the tofu dependencies and threw errors at every try

=> It appears we forgot to include a **requirements.txt** file that pip uses to check the dependencies
=> see this [site ](https://devcenter.heroku.com/articles/python-pip#the-basics) for details on pip dependencies managing

To try and reproduce : create an empty ne conda package on a mac (with python 3.7, including just the basics like conda, pip). 
Then try installing tofu using pip install tofu

=> it should complain about missing dependencies (due to missing requirements.txt)
=>and Ricardo also has an issue because he doesn't have gcc-8 (is it the only solution ? he has another compiler called LLVM)

Also, it seems we forgot **openmp** in the dependencies for installation

rragona, 2019-11-15:
Thanks @Didou09 for your help and for opening this issue.
This is the error I have while trying:
```
pip install tofu

Collecting tofu
  Using cached https://files.pythonhosted.org/packages/5a/06/cbb85978151b0854bd2d2f022cfad1ff15a77db88593aeb236cf671bcd89/tofu-1.4.1.tar.gz
    ERROR: Command errored out with exit status 1:
     command: /Users/rik/anaconda/envs/tofu/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/51/1mbjm0lj6098n3_rv0g7tfdr0000gn/T/pip-install-lw_tfc9_/tofu/setup.py'"'"'; __file__='"'"'/private/var/folders/51/1mbjm0lj6098n3_rv0g7tfdr0000gn/T/pip-install-lw_tfc9_/tofu/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /private/var/folders/51/1mbjm0lj6098n3_rv0g7tfdr0000gn/T/pip-install-lw_tfc9_/tofu/pip-egg-info
         cwd: /private/var/folders/51/1mbjm0lj6098n3_rv0g7tfdr0000gn/T/pip-install-lw_tfc9_/tofu/
    Complete output (18 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/51/1mbjm0lj6098n3_rv0g7tfdr0000gn/T/pip-install-lw_tfc9_/tofu/setup.py", line 151, in <module>
        openmp_installed = not check_for_openmp(os.environ['CC'])
      File "/private/var/folders/51/1mbjm0lj6098n3_rv0g7tfdr0000gn/T/pip-install-lw_tfc9_/tofu/setup.py", line 139, in check_for_openmp
        stdout=fnull, stderr=fnull)
      File "/Users/rik/anaconda/envs/tofu/lib/python3.7/subprocess.py", line 339, in call
        with Popen(*popenargs, **kwargs) as p:
      File "/Users/rik/anaconda/envs/tofu/lib/python3.7/subprocess.py", line 800, in __init__
        restore_signals, start_new_session)
      File "/Users/rik/anaconda/envs/tofu/lib/python3.7/subprocess.py", line 1551, in _execute_child
        raise child_exception_type(errno_num, err_msg, err_filename)
    FileNotFoundError: [Errno 2] No such file or directory: 'gcc-8': 'gcc-8'
    cython version = 0.29.13
    numpy  version = 1.17.3
    cython version = /Users/rik/anaconda/envs/tofu/lib/python3.7/site-packages/Cython/__init__.py
    numpy  version = /Users/rik/anaconda/envs/tofu/lib/python3.7/site-packages/numpy/__init__.py
    ................ checking if openmp installed...
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

lasofivec, 2019-11-15:
Hey @rragona 
I added a `requirements.txt` for the release that's coming in the next few days.
Also the error regarding `gcc-8` should be solved. 
But if you want we can chat a bit on how to make it work in the meantime: https://gitter.im/ToFuProject/community

flothesof, 2019-11-15:
Hi guys,

Just a thought, but as far as I understand the machinery, requirements.txt is not necessary. 

Why? Because setup.py already includes dependencies.

As far as I can tell, the error is due to the heuristic that checks for the existence of the compiler and not the dependencies. 

My guess is that this will not solve any problems but adding the file will confuse us all... 
Sorry for the strong opinion, but I really believe this has nothing to do with requirements.

flothesof, 2019-11-15:
Okay, so after reading this again, I have to take some of what I said back. What is the current status of Mac OS as a platform? Are we not recommending to install from source in the docs? And not from pypi?

lasofivec, 2019-11-15:
> As far as I can tell, the error is due to the heuristic that checks for the existence of the compiler and not to the dependencies.

I talked directly with Didier, apparently they first tried a `pip install` that failed because it couldn't find the `requirements.txt` file. They installed this by hand and then got the error mentioned above, which does come from an error from my part in the `setup.py` that is now corrected in the `devel` branch. In short: there were 2 problems when installing with pip on Mac, the error message has nothing to do with the `requirements` file

> Are we not recommending to install from source in the docs? 

Yes, I also mentioned this to @Didou09. But it seems like installing directly from pip should also be a possibility.

I should also add the `requirements.txt` is quite standard and updates in github the dependencies. So I see no problem in adding it.

flothesof, 2019-11-15:
I'm still thinking something about this is weird. In some of my other projects, I've defined dependencies in the setup.py and made wheels that registered the right dependencies. There was no need to add requirements.txt. I'm puzzled about why it is necessary here. Do you have any explanation why it works with it?

The disadvantage of adding this file I see is that there are now two places where requirements are defined. Hence when changing these, we will run into the risk of forgetting to update. It adds easy to miss complexity. Couldn't it be that it's a false positive?

Final thing I don't understand: pip install did not work at first but by adding local files it started to work? This is a weird way of installing: neither pure pip nor pure from source. For me it looks hacky / I'm thinking I'm missing something.

flothesof, 2019-11-16:
So, for the record, I have symptoms when installing from scratch under MacOS, linked to the compiler for the Cython parts.
Following https://tofuproject.github.io/tofu/installation.html#installing-as-a-developer, I have noticed two things:

- `pip install -e .[dev]` does not work in my shell by `pip install -e ".[dev]"` does
- it hangs due to gcc throwing an error, which was first resolved by running `xcode-select --install` but still fails later due to this:

tofu/geom/_vignetting_tools.cpp:610:10: fatal error: 'ios' file not found
    #include "ios"

So as of right now, I'm unable to install tofu on my Mac...

Didou09, 2019-11-16:
So:

- What was done: empty conda environment, installing from source (pip). It failed because pip could not find the requirements.txt and dependencies were missing. So dependencies were installed by hand (conda install) and finally pip was retried, it failed this time because of the error in the setup.py, now fixed, that Laura mentioned.

- What you are saying @flothesof is that pip does not need the requirements.txt when you create a pip wheel. But this was not a pip wheel. It was installing from source using pip. We never created pip wheels. Maybe we should though ? (separate issue in that case)

- I agree storing dependencies in separate places is a source of complexity. But it is already the case. The dependencies are in the setup and also in the conda recipe (for creating the conda package).

- I feel that, by principle, installing from source with pip should always be possible and easy. What do you think ?

=> So, although it does introduce a bit of extra complexity, it does seem useful to add the requirements.txt as Laura did for non-developpers who wnat to install on Mac or windows from source using pip. 
Complexity is an issue, but tofu being a difficult to install for newbies is a higher-priority issue. If requirements.txt make it easier for them I feel it is the way to go.
Furthermore, the requirements.txt being very short and simple could be automatically generated through Continuous Integration in a near future, what do you think ?

flothesof, 2019-11-16:
Status report after discussing this with @Didou09:

- The goal is to get pip install tofu to work on Mac 
- there is a distinction between build requirements (Cython, numpy which are imported by setup.py and therefore necessary to install from source) and user requirements (others already present in setup.py, later user in tofu itself like matplotlib, scipy)
- adding requirements.txt seems to solve the build requirements issue (on @lasofivec 's machine) but it's not clear why  
- a better hotfix would be to upload a wheel to PyPI obtained using `python setup.py bdist_wheel`

As a side note, I was able to compile from source on Mac without requirements.txt, which I interpret to show that we don't need it...

Didou09, 2019-11-16:
Thanks for the status report, 
so we go for both (requirements.txt + bdist wheel) for the next release to try and make this as robust as possible.
We'll see later whether we can afford to remove the requirements.txt to reduce the complexity

flothesof, 2019-11-17:
I've just tested the wheel build. On my machine, I am able to:
- create a .whl file in a Py36 env
- create a new test conda environment without any dependencies except pip (also Py36) and pip install path/to/.whl
- I am then able to run python and import tofu

Another test would be that I send the .whl I've built to @lasofivec to make sure this also works.

Didou09, 2019-11-18:
Yes, building a wheel is a good complemnetary to adding the requirements.txt (ceinture + bretelles). If it works, we can upload it to Pypi.

Could you by any chance try to build a conda package for Mac ? (low priority, but just in case)

Didou09, 2019-11-18:
Also, I saw that the OSX platform seems to have been added recently to the Travis CI available platforms. 
@lasofivec , do you think you could have a look at [this link](https://docs.travis-ci.com/user/multi-os/#operating-system-differences) and see if we can create a build Matrix in the .travis.yml that would also build on Mac OS ?

lasofivec, 2019-11-19:
Ok I'm sorry I'm arriving a bit late to the conversation... 

> Final thing I don't understand: pip install did not work at first but by adding local files it started to work? 

It wasn't adding files but installing the "requirements" (dependencies) "by hand". As Didier explained.

> Following https://tofuproject.github.io/tofu/installation.html#installing-as-a-developer, I have noticed two things:

I think you (@flothesof) re-wrote the doc for Mac. When the doc was in tabs form, it was written by me and it wasn't at all the same procedure. As you are more familiar with packagings, I left it unchanged. We will have to update the doc in the next release whichever way we decide it's the best on Mac.
Installing the Xcode utilities was in the original version of the doc (I think).
I don't get the same bug, if you want we can see what's different between our configurations, but I would open a different issue for this (I'm pretty sure is related to Mojave).

lasofivec, 2019-11-19:
> Status report after discussing this with @Didou09:
>
> * The goal is to get pip install tofu to work on Mac
> * there is a distinction between build requirements (Cython, numpy which are imported by [...]

Yes, I added scipy and matplotlib since they are quickly needed for users and they match the requirements in the `setup.py`
I am not sure if you mentioned the difference so that we take the libraries only needed when using tofu out of the `requirements.txt`, or if you want me to write them in a separate category on the file, or just as a "FYI" ?
 
> * adding requirements.txt seems to solve the build requirements issue (on @lasofivec 's machine) but it's not clear why

I dont understand why is there a confusion ? "pip install tofu" without the requirements file doesn't work because it says there are dependencies missing, when you tell pip which dependencies are missing (with the use of `requirements.txt`), it install them, then installs tofu... 
I think you should also be able to try it with the version with and without the file.

> * a better hotfix would be to upload a wheel to PyPI obtained using `python setup.py bdist_wheel`
> 
> As a side note, I was able to compile from source on Mac (`CFLAGS='-stdlib=libc++' pip install -e ".[dev]"`) without requirements.txt, which I interpret to show that we don't need it...

"from source" but also with an environment from scratch ? tofu was always installable if you already had the right dependencies in you environment/machine. 

ok for the wheel :)

lasofivec, 2019-11-19:
> I've just tested the wheel build. On my machine, I am able to:
>
> * create a .whl file in a Py36 env
> * create a new test conda environment without any dependencies except pip (also Py36) and pip [...]

As we talked by email, I obtained an error during installation that didn't prevent the installation of tofu from happening but that for me remains unexplained. 

`ERROR: pytools 2018.5.2 requires decorator>=3.2.0, which is not installed.
`

For the moment I would suggest keeping both suggestions, but I trust you judgment. If you want to get rid of the `requirements.txt` and believe the wheel is robust enough, I agree! :)

lasofivec, 2019-11-19:
> Also, I saw that the OSX platform seems to have been added recently to the Travis CI available platforms.
> @lasofivec , do you think you could have a look at [this [...]

this is already an issue somewhere else and, in my opinion, separated from this one :) 
but yes

Didou09, 2019-11-19:
The requirements.txt was added, in devel, by PR #251 

It is a hotfix, but the outcome of this discussion is that we should probably discuss all these aspects (CI / installation / deployments on various platforms) in more details in a near future. And open a dedicated issue or even project / milestone.

lasofivec, 2019-11-20:
Adding to this conversation just FTR, I recalled why the compiler used with OS X was set manually:

https://github.com/ToFuProject/tofu/issues/183

So @rragona It will be good if you are sure to use the latest `gcc` version available for your distribution.

rragona, 2019-11-22:
Just a small update on this topic.
I have set up a clean conda environment on OS X:
`conda create --name tofu python=3.7 ipython -y`
`source activate tofu`

I have downloaded `requirement.txt` from the devel branch and executed:
`pip install -r requirements.txt tofu`

I have now a working tofu!
```
Python 3.7.5 (default, Oct 25 2019, 10:52:18) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.9.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import tofu as tf                                                                                                                                                               
~/anaconda/envs/tofu/lib/python3.7/site-packages/tofu/imas2tofu/__init__.py:27: UserWarning: 

IMAS python API issue
imas could not be imported into tofu ('import imas' failed):
  - it may not be installed (optional dependency)
  - or you not have loaded the good working environment

    => the optional sub-package tofu.imas2tofu is not usable

  warnings.warn(msg)
~/anaconda/envs/tofu/lib/python3.7/site-packages/tofu/mag/__init__.py:28: UserWarning: Traceback (most recent call last):
  File "~/anaconda/envs/tofu/lib/python3.7/site-packages/tofu/mag/__init__.py", line 11, in <module>
    import imas
ModuleNotFoundError: No module named 'imas'

    => the optional sub-package tofu.mag is not usable

  warnings.warn(msg)

In [2]:
```

I have also repeated in an another clean environment and this procedure works smoothly.

CAVEAT: Following the suggestion of @lasofivec I have installed `gcc-8` from homebrew.

lasofivec, 2019-11-22:
Awesome :) thanks for the report ! we are currently working with Didier on how to make it easier !

closed 2020-04-07 by Didou09