ToFuProject/tofu #255: Possible bug / inconsistency in _GG.LOS_calc_signal()

Didou09, 2019-11-18:
While trying to finish / polish the PR #254 , and testing all tutorials, an error occured for:

tutorial plot_custom_emissivity.py
minimize = 'hybrid'
method = 'sum'

Upon closer inspection, it comes from line 3132 in _GG.py.
It seems method = 'sum' is the only case where the result is stored in sig_mv instead of sig.
I suspect this is the source of the error..
Indeed, there is no error if method  = 'simps' or method = 'romb'

The error-check matrix is

|      | sum | simps | romb |
|----|------|--------|-------|
hybrid | Error | ok | ok
calls | ok | ok | ok
memory | ok | ok | ok

The error message is:
```
Traceback (most recent call last):
  File "/Applications/Anaconda/python36/lib/python3.6/site-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/Home/DV226270/ToFu_All/tofu_git/tofu/tofu/tests/tests09_tutorials/tests01_runall.py", line 182, in test03_plot_custom_emissivity
    self._test_tuto('tuto_plot_custom_emissivity')
  File "/Home/DV226270/ToFu_All/tofu_git/tofu/tofu/tests/tests09_tutorials/tests01_runall.py", line 164, in _test_tuto
    raise Exception(msg)
Exception: failed to get the current screen resources
Traceback (most recent call last):
  File "/Home/DV226270/ToFu_All/tofu_git/tofu/tuto_plot_custom_emissivity.py", line 92, in <module>
    t=time_vector)
  File "/Home/DV226270/ToFu_All/tofu_git/tofu/tofu/geom/_core.py", line 6066, in calc_signal
    Test=True,
  File "tofu/geom/_GG.pyx", line 3132, in tofu.geom._GG.LOS_calc_signal
TypeError: only size-1 arrays can be converted to Python scalars
```

@lasofivec can you have a look on a dedicated branch ?

Didou09, 2019-11-18:
Important update:

When running the tutorial, I realize the error seems to be linked to the way the emissivity function is defined:

instead of using emissivity as input function we use:
`ff = lambda pts, t=None: np.zeros((1, pts.shape[1])) +1.0 if t is None else np.zeros((len(t), pts.shape[1])) +1.0
`
There is no error showing up...

What is strange is that both function are defined correctly, their output have the same shape:

```
In [31]: time_vector = np.linspace(0, 2 * np.pi, num=100)

In [32]: pts = np.random.random((3,10))

In [33]: emissivity(pts).shape, emissivity(pts, t=time_vector).shape
Out[33]: ((1, 10), (100, 10))

In [34]: ff(pts).shape, ff(pts, t=time_vector).shape
Out[34]: ((1, 10), (100, 10))
```
 But the error only shows up when using emissivity...

I suspect it might have something to do with the output being C-contiguous ?
@lasofivec could you try to investigate more closely, and if necessary (i.e. if there are additional constraints of the user-function), include them in the new error message in check_ff() ? (in the full error message)

lasofivec, 2019-11-18:
Could you also plot the C-contiguous flag of the arrays ?

Didou09, 2019-11-19:
Fixed in devel by PR #257

PR #257 (2019-11-19): [bug fix #255] found bug
Fixes #255
commit 11ab20c2 (2019-11-19): [bug fix #255] found bug
commit 8b88511d (2019-11-19): Merge pull request #257 from ToFuProject/bf-#255

[bug fix #255] found bug
closed 2019-11-20 by Didou09