ToFuProject/tofu #271: Installation issues of tofu 1.4.2-alpha

Didou09, 2019-11-21:
Several issues were found on various platforms durign installation or during import time:

All errors occur at import time

| Case          |    1                               |       2
|-------------|-------------------------|------------
| OS             | Ubuntu 16.04.6 (Atlas) | Ubuntu 16.04.6 (Atlas)
| gcc            | 6.4.0                             | 6.4.0
| GLIBC        | 2.23                               |  2.23 
| python      | 3.6.9 (anaconda)           | 3.7
| conda        | 4.5.12                            | 
| numpy      |  1.17                              |
| np error    |                                       |
| matplotlib | 3.1.1                              |
| mpl error  | (numpy)                         |
| tofu          |                                       |
| tofu err    |  (numpy)                          |

ERRATUM
-----------
No problem on ATLAS, the libraries that were being used had the wrong version (installed with pip previously in ~usr/local).

lasofivec, 2019-11-21:
Procedure to find bug was:

- creating a new (empty) conda environment
- checking python version and path `$ python --version` and `$ which python` (should be in the environment)
- installing faulty packages one by one (`$ conda install <package>`)
- checking that their version and path were right, eg. `$ python -c "import numpy as np; print(np.__version__); print(np.__file__)`

closed 2019-11-21 by lasofivec

Didou09, 2019-11-21:
The problem with creating the package and with import wrong matplotlib version was simply the existence of an alias refering to a fixed python version => destroy the alias solves everything