ToFuProject/tofu #272: pip install fails due to case-sensitivity for Cython vs cython

Didou09, 2019-11-21:
Starting from a quasi empty conda env, I try to install tofu with pip.

I get an error about Cython, from the setup.py.
**C**ython is not known, because **c**ython is listed in the dependencies.

Note: installing with pip into a conda env works only if we use the pip that installed by conda into that same env (check before running pip install):

```
(base) [ DV226270 spica ~ ] conda create -n empty python=3.6 pip ipython
(base) [ DV226270 spica ~ ] conda activate empty
(empty) [ DV226270 spica ~ ] which pip
~/.conda/envs/empty/bin/pip
(empty) [ DV226270 spica ~ ] pip install tofu
Collecting tofu
  Downloading https://files.pythonhosted.org/packages/5a/06/cbb85978151b0854bd2d2f022cfad1ff15a77db88593aeb236cf671bcd89/tofu-1.4.1.tar.gz (19.6MB)
     |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 19.6MB 12.6MB/s 
    ERROR: Command errored out with exit status 1:
     command: /Home/DV226270/.conda/envs/empty/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-aptzshyh/tofu/setup.py'"'"'; __file__='"'"'/tmp/pip-install-aptzshyh/tofu/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-aptzshyh/tofu/pip-egg-info
         cwd: /tmp/pip-install-aptzshyh/tofu/
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-aptzshyh/tofu/setup.py", line 14, in <module>
        import Cython as cth
    ModuleNotFoundError: No module named 'Cython'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

In ipython, importing cython and Cython does not make a difference (except the file from which they are loaded, Cython is the __init__.py, cython is cython.py).
=> Replace all Cython occurences by cython

Didou09, 2019-11-21:
Doesn't pass tests... Cython is not a package 

=> maybe the other way around ? Cython in requirements and setup instead of cython ?

lasofivec, 2019-11-21:
> Doesn't pass tests... Cython is not a package

"Cython" is not a package or "cython" ?

lasofivec, 2019-11-21:
Ok this is normal... Related to issue #250 !! you are installing the old tofu version !!

Didou09, 2019-11-21:
OK I tried the other way around, it works for conda, but I need to create a pip sdist and try to install it localy to be sure. Will do it tomorrow morning

lasofivec, 2019-11-21:
you can try to install the pre release version `` pip install --pre tofu ``

lasofivec, 2019-11-21:
https://github.com/coleifer/unqlite-python/issues/30

I finally understood what's going on: setup.py needs python (and numpy) thus the installation always fail.
https://github.com/coleifer/unqlite-python/issues/30

Didou09, 2019-11-22:
Thanks,

I tried introducing cython and numpy in the setup_requires, but it still fails with the same error.

I think it is because it needs to import cython when first importing the setup (and before reaching the setup_requires line).
It would need first to import the setup without cython, read the setup_require line, install cython and then run the setup() function.
A workaround is proposed here:
https://stackoverflow.com/questions/37471313/setup-requires-with-cython

lasofivec, 2019-11-22:
Why dont you leave Cython/cython as it was, and just add the setup_requires (i did this in a branch)
it's impossible it will fail...

Didou09, 2019-11-22:
I added the setup_requires,still fails... can you take over this branch from here ? I have a meeting in 15 min.

I have edited the setup.py to define all necessary workarounds, as described in the link, plus added a few comments (questions)

After pushing:
* create a sdist: python setup.py sdist
     => will create a sdist in tofu/dist/tofu-version...
* create an empty environment with just python and pip: conda create -n empty python=3.6 pip
* activate that environment
* try installing tofu with pip from the local sdist directly. It should work and handle all dependencies

Doesn't work yet...

Didou09, 2019-11-22:
I can confirm, after local a test, that hord-coding the numpy.get_include() path solves the problem.
The only remaining problem is the import numpy that happens before the call to setup()

Didou09, 2019-11-22:
This is a good description of our problem:
https://stackoverflow.com/questions/54117786/add-numpy-get-include-argument-to-setuptools-without-preinstalled-numpy

Didou09, 2019-11-22:
Opened a corresponding issue on setuptools:
https://github.com/pypa/setuptools/issues/1910

Didou09, 2019-11-22:
That's probably the way to go, what do you think (before calling the setup() function):

```
dist.Distribution().fetch_build_eggs(['Cython>=0.15.1', 'numpy>=1.10'])
```
from:
https://luminousmen.com/post/resolve-cython-and-numpy-dependencies

https://github.com/pypa/pip/issues/5761

Didou09, 2019-11-22:
The setuptools guys answered the opened issue and rediected towards:
https://www.python.org/dev/peps/pep-0518/

https://www.python.org/dev/peps/pep-0517/

closed 2019-11-22 by Didou09

lasofivec, 2019-11-22:
> That's probably the way to go, what do you think (before calling the setup() function):
> 
> ```
> dist.Distribution().fetch_build_eggs(['Cython>=0.15.1', 'numpy>=1.10'])
> ```
> 
> from:
> https://luminousmen.com/post/resolve-cython-and-numpy-dependencies
> 
> [pypa/pip#5761](https://github.com/pypa/pip/issues/5761)

I tried this, and it didn't work on my case..

Didou09, 2019-11-22:
Ok, so only PEPs are ledt as a solution, crossing fingers...