ToFuProject/tofu #276: setup.py build dependencies: numpy/cython and PEP recommendations

Didou09, 2019-11-22:
As explained in the last comments of issue #272 :
* pip install runs setup.py
* setup.py, in tofu, needs numpy and cython at import time, which means the user would need to install them by hand before running pip install. We want to avoir that, pip should handle all dependencies (run and build)
* The recommended modern solution is PEP517 and PEP518, which implies creating a tofu.toml file list build-time dependencies with proper formatting
* The recommended solution is applicable since pip >= 18.0

To be clarified : should we still include setup(setup_requires=['cython', 'numpy']) in addition to the tofu.toml file ? 

Ressources:
* Issue #272 
* Hacky solutions before PEP517 and PEP518: 
   - https://stackoverflow.com/questions/19919905/how-to-bootstrap-numpy-installation-in-setup-py/21621689#21621689
    - https://stackoverflow.com/questions/54117786/add-numpy-get-include-argument-to-setuptools-without-preinstalled-numpy
    - https://stackoverflow.com/questions/21605927/why-doesnt-setup-requires-work-properly-for-numpy/21621493#21621493
* [Opened issue](https://github.com/pypa/setuptools/issues/1910) on setuptools
* [PEP517](https://www.python.org/dev/peps/pep-0517/#build-requirements) and [PEP518](https://www.python.org/dev/peps/pep-0518/#specification)

Didou09, 2019-11-26:
Presumably fixed, in devel, by PR #283  (to be validated by users on tofu 1.4.2a3)

flothesof, 2020-05-07:
Ping @Didou09 @lasofivec I think this issue is largely fixed too by the introduction of the pyproject.toml file. Candidate for closing :)

closed 2020-05-07 by Didou09