ToFuProject/tofu #287: Check imas version and warn user

Didou09, 2019-11-27:
On the ITER cluster, and on any install where IMAS is loaded via linux environment modules, imas2tofu should check, at import time, the currently loaded imas version against the latest version available.

If a more recent version is available, it should issue a warning message accordingly

lasofivec, 2019-11-27:
This looks like an IMAS issue to me ?

Didou09, 2019-11-27:
It is a user-friendly feature that just checks which module was loaded in the user environment against the other modules available (i.e.: module list vs module avail).
I noticed that I tend to pre-load a list of modules on the ITER cluster and only rarely check that they are up-to-date.
I suspect other users do the same.
But the IMAS versions change rapidly, new modules are created quite often with easyconfig, and the user may not be aware that he has loaded an old version of IMAS.

This will warn him :-)

Didou09, 2019-11-27:
Fixed, in devel, by PR #289

lasofivec, 2019-11-27:
Do you do `module load <package>=<version>` if you want to always use the latest version you can just `module load <package>`.
I will accept the PR but it seems to me something that tofu shouldn't (have to) handle

Didou09, 2019-11-27:
I don't do module load, tofu should not load / unload packages.
I just check the already loaded packages (module list vs module avail) and inform the user.

It is done in imas2tofu, a sub-package dedicated to the imas interfacing.
It imas is not present, imas2tofu is not loaded and this is done anyway

lasofivec, 2019-11-28:
I meant what you do in your `bashrc` I agree this has nothing to do with tofu ;)

closed 2020-04-07 by Didou09