ToFuProject/tofu #295: test failing on Windows

flothesof, 2019-11-27:
After installing from source, I'm running nosetests. 

Everything runs, except for one unit test.

```
  File "C:\Users\FL232714\Documents\GitHub\tofu\tofu\tests\tests09_tutorials\tests01_runall.py", line 20, in <module>
    _TFROOT = _HERE[:-_HERE[::-1].index('/tofu'[::-1])-len('/tofu')]
ValueError: substring not found
```

Will try to fix.

flothesof, 2019-11-27:
It fails due to the assumption that path delimiters are the same on all platforms. 
[image]

Didou09, 2019-11-28:
Fixed, in devel by PR #296

flothesof, 2019-12-04:
Can confirm all tests run from devel 2d9b1837a126d4ad9f88b8b7779ab5ddd58a5f66.

Didou09, 2019-12-04:
Note for the future:
We générally don't close issues once a PR on develk fixes them, we simply tag them as "Fixed in devel"

Motivations:
* Issues are finally closed only when they are solved in the last official (non-alpha) release. Until then they are still open for users who are not using a development version
* It makes life easier for us to summarize the changes in a new version. We simply collect all isues labelled "Fixed in devel" to write the changelog. And we close them after the release

PR #296 (2019-11-28): Fix issue295
Fixes #295.
closed 2020-04-07 by Didou09