ToFuProject/tofu #310: Creating camera fails

flothesof, 2019-12-03:
When I run the following example code from devel 2d9b1837a126d4ad9f88b8b7779ab5ddd58a5f66:

```python
import matplotlib.pyplot
import numpy as np
import tofu as tf

config = tf.geom.utils.create_config()

cam2d = tf.geom.utils.create_CamLOS2D(
    config=config,
    P=[3.4, 0, 0],
    N12=100,
    F=0.1,
    D12=0.1,
    angs=[np.pi, np.pi/6, 0],
    Name="",
    Exp="",
    Diag="",
)
```

I get the following error:
``` 
  File "C:\Users\FL232714\Documents\GitHub\tofu\tofu\geom\utils.py", line 1011, in create_CamLOS2D
    return _create_CamLOS(nD=2, **kwdargs)
  File "C:\Users\FL232714\Documents\GitHub\tofu\tofu\geom\utils.py", line 1000, in _create_CamLOS
    color=color, config=config, SavePath=SavePath)
  File "C:\Users\FL232714\Documents\GitHub\tofu\tofu\geom\_core.py", line 3607, in __init__
    super(Rays, self).__init__(**kwdargs)
  File "C:\Users\FL232714\Documents\GitHub\tofu\tofu\utils.py", line 1388, in __init__
    self._init(**kwdargs)
  File "C:\Users\FL232714\Documents\GitHub\tofu\tofu\geom\_core.py", line 4033, in _init
    self._set_dgeom(sino=True, **kwdgeom)
  File "C:\Users\FL232714\Documents\GitHub\tofu\tofu\geom\_core.py", line 4588, in _set_dgeom
    self.compute_dgeom(extra=extra)
  File "C:\Users\FL232714\Documents\GitHub\tofu\tofu\geom\_core.py", line 4475, in compute_dgeom
    Nstep=50,
  File "C:\Users\FL232714\Documents\GitHub\tofu\tofu\geom\_plot.py", line 787, in _LOS_calc_InOutPolProj_Debug
    fs=fs, wintit=wintit, draw=False)
  File "C:\Users\FL232714\Documents\GitHub\tofu\tofu\geom\_core.py", line 3292, in plot
    Test=Test,
  File "C:\Users\FL232714\Documents\GitHub\tofu\tofu\geom\_plot.py", line 198, in Struct_plot
    draw=False, **dplot[proj])
  File "C:\Users\FL232714\Documents\GitHub\tofu\tofu\geom\_plot.py", line 473, in _Plot_3D_plt_Ves
    ax = _def.Plot_3D_plt_Tor_DefAxes(fs=fs, wintit=wintit)
  File "C:\Users\FL232714\Documents\GitHub\tofu\tofu\geom\_def.py", line 167, in Plot_3D_plt_Tor_DefAxes
    ax.set_aspect(aspect="equal", adjustable='datalim')
  File "C:\Anaconda3\envs\tofu_dev\lib\site-packages\matplotlib\axes\_base.py", line 1281, in set_aspect
    'It is not currently possible to manually set the aspect '
NotImplementedError: It is not currently possible to manually set the aspect on 3D axes
``` 

Is this normal? 
I expect the camera creation to work in this case (maybe that's a wrong assumption).

Thanks,
Florian

Didou09, 2019-12-03:
File "C:\Users\FL232714\Documents\GitHub\tofu\tofu\geom\_plot.py", line 787, in _LOS_calc_InOutPolProj_Debug
    fs=fs, wintit=wintit, draw=False)

This line shows that with the camera setup you provided, some of the LOS are outside of the vessel (ie. some pixels of the camera see nothing, maybe because your camera is too far out).

tofu tries to plot a debug 3d graph to show you these LOS so you can better correct.
But the debug plot itself fails... (not touched in quite a while.

=> Adjust your camera position
=> I'll do a PR to make this more robust with a try / except and a more explicit error message
Thanks for reporting !

Didou09, 2019-12-03:
Fixed, in devel, by PR #311

Didou09, 2021-02-16:
Closed since... probably a while or maybe to the numerous improviements and debugging we've done since, like PR #465 #464 #456,

closed 2021-02-16 by Didou09