ToFuProject/tofu #312: Setting default repo branch to devel

flothesof, 2019-12-03:
Hi,

this is just a small setting.
When I create new branches with the Github app, it does not use devel as the default branch.
However, this can be set easily by following this tutorial: https://help.github.com/en/github/administering-a-repository/setting-the-default-branch

Can you set **devel** as the default branch for the tofu repo?

Thanks
Florian

Didou09, 2019-12-03:
Done, closing this issue
Thanks

closed 2019-12-03 by Didou09

lasofivec, 2019-12-04:
There is only one thing that bothers me when doing so: the first source code you get when you go to github.com/ToFuProject/tofu now is the source code in devel.  But this is not the code you will get when you install tofu...
It is a matter of convention anyhow, so however you prefer. But it's something to keep in mind

Didou09, 2019-12-04:
Good point
Given the efforts we made to make tofu available through conda and pip on several platforms, I suggest:
* users install it from conda / pip, they get the last stable version
* cloning from github is for contributors / advanced users who want the latest features, so it makes sense for them to get the devel branch (which is stable too)

What do you think ?