ToFuProject/tofu #314: Test02_DataCam12DSpectral failing randomly on mac os X

lasofivec, 2019-12-04:
Since trying the deployment in OSX platforms, from time to time I get the following test failing.
Not sure why/how, I will investigate this more when the deployment branch is done.
If in the mean time this rings a bell (@Didou09) I would gladly appreciate the help.

> ERROR: test suite for <class 'tofu.tests.tests02_data.tests03_core.Test02_DataCam12DSpectral'>
> ----------------------------------------------------------------------
> Traceback (most recent call last):
>   File "/Users/travis/miniconda/lib/python3.7/site-packages/nose/suite.py", line 210, in run
>     self.setUp()
>   File "/Users/travis/miniconda/lib/python3.7/site-packages/nose/suite.py", line 293, in setUp
>     self.setupContext(ancestor)
>   File "/Users/travis/miniconda/lib/python3.7/site-packages/nose/suite.py", line 316, in setupContext
>     try_run(context, names)
>   File "/Users/travis/miniconda/lib/python3.7/site-packages/nose/util.py", line 470, in try_run
>     return func()
>   File "/Users/travis/build/ToFuProject/tofu/tofu/tests/tests02_data/tests03_core.py", line 432, in setup_class
>     conf1 = tfg.utils.create_config(case='B3')
>   File "/Users/travis/build/ToFuProject/tofu/tofu/geom/utils.py", line 831, in create_config
>     returnas=returnas, path=path)
>   File "/Users/travis/build/ToFuProject/tofu/tofu/geom/utils.py", line 748, in _create_config_testcase
>     out=returnas)
>   File "/Users/travis/build/ToFuProject/tofu/tofu/geom/_core.py", line 1728, in from_txt
>     color=color,
>   File "/Users/travis/build/ToFuProject/tofu/tofu/geom/_core.py", line 218, in __init__
>     super(Struct, self).__init__(**kwdargs)
>   File "/Users/travis/build/ToFuProject/tofu/tofu/utils.py", line 1388, in __init__
>     self._init(**kwdargs)
>   File "/Users/travis/build/ToFuProject/tofu/tofu/geom/_core.py", line 603, in _init
>     self._set_dgeom(**kwdgeom)
>   File "/Users/travis/build/ToFuProject/tofu/tofu/geom/_core.py", line 642, in _set_dgeom
>     Clock=Clock,
>   File "/Users/travis/build/ToFuProject/tofu/tofu/geom/_comp.py", line 66, in _Struct_set_Poly
>     assert Vol > 0.0, msg
> AssertionError: Pb. with volume computation for Ves object of type 'Tor' !

Didou09, 2019-12-04:
This is a safety test checking that the volume of the torus is > 0.
If it fails it means that it may be negative, which could indicate a clock-wise (instead of counter-clockwise) polygon.
Or the volume is 0 or nan for some reason.

But it is strange that it happens only on Mac and randomly. If there was a systematic error we should have seen it earlier.

Could you try a step-by-step debugging on a Mac ? Maybe identify the polygon and see if the volume could be very small (i.e. below rounding errors ?)

lasofivec, 2019-12-04:
it has never happened on my mac, I have trie a couple of times. But it's hard to say if it hasn't happened because it will not happen or if I just haven't tried enough times :) 
Will let you know asap

lasofivec, 2020-01-09:
Looks that it was coming from `Poly_isClockwise` (in `GG`):

- The modulo operator in Cython doesn't give the right results when `@cython.cdvision(True)` is set.

Didou09, 2020-01-09:
Fixed, in devel, by PR #323

closed 2020-04-07 by Didou09