ToFuProject/tofu #326: TypeError in cython file when compiled on CEA cluster

Didou09, 2020-01-13:
Description:
-------------

On CEA linux servers, if I get the last update on devel (git pull origin devel) and compile locally (python setup.py build_ext --inplace), the compilation works fine.

But when I open an ipython console and try to import tofu I get a TypeError in on eof the cythin files, referring to the signature of a function in _basic_geom_tools.pyx:

```
In [1]: import tofu as tf
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
/Home/DV226270/ToFu_All/tofu_git/tofu/tofu/geom/_core.py in <module>()
     24 try:
---> 25     import tofu.geom._def as _def
     26     import tofu.geom._GG as _GG

AttributeError: module 'tofu' has no attribute 'geom'

During handling of the above exception, another exception occurred:

TypeError                                 Traceback (most recent call last)
<ipython-input-1-d22734232f74> in <module>()
----> 1 import tofu as tf

/Home/DV226270/ToFu_All/tofu_git/tofu/tofu/__init__.py in <module>()
     66 from tofu.utils import save, load, load_from_imas, calc_from_imas
     67 import tofu._plot as _plot
---> 68 import tofu.geom as geom
     69 import tofu.data as data
     70 

/Home/DV226270/ToFu_All/tofu_git/tofu/tofu/geom/__init__.py in <module>()
     10 """
     11 
---> 12 from tofu.geom._core import PlasmaDomain, Ves, PFC, CoilPF, CoilCS, Config
     13 from tofu.geom._core import Rays, CamLOS1D, CamLOS2D
     14 from tofu.geom._core_optics import *

/Home/DV226270/ToFu_All/tofu_git/tofu/tofu/geom/_core.py in <module>()
     29 except Exception:
     30     from . import _def as _def
---> 31     from . import _GG as _GG
     32     from . import _comp as _comp
     33     from . import _plot as _plot

/Home/DV226270/ToFu_All/tofu_git/tofu/tofu/geom/_GG.pyx in init tofu.geom._GG()
----> 1 # cython: boundscheck=False
      2 # cython: wraparound=False
      3 # cython: initializedcheck=False
      4 # cython: cdivision=True
      5 #

TypeError: C function tofu.geom._basic_geom_tools.compute_hypot has wrong signature (expected arrayobject *(__Pyx_memviewslice const , __Pyx_memviewslice const , struct __pyx_opt_args_4tofu_4geom_17_basic_geom_tools_compute_hypot *__pyx_optional_args), got arrayobject *(__Pyx_memviewslice, __Pyx_memviewslice, struct __pyx_opt_args_4tofu_4geom_17_basic_geom_tools_compute_hypot *__pyx_optional_args))
```

Context:
---------
CentOS 6.10
ipython 5.1.0
python 3.6.9
cython 0.26.1
gcc 6.4.0
commit e6c9f42b93ade41f24e906da705df4847dc54ee3

Attached is a txt file containing the output of the compilation command python setup.py build_ext --inplace
[cython.txt](https://github.com/ToFuProject/tofu/files/4052861/cython.txt)

Didou09, 2020-01-13:
Correction:
------------

This issue disappears if I do a clean compilation:

`python setup.py clean build_ext --inplace`

I am closing it

closed 2020-01-13 by Didou09

lasofivec, 2020-01-21:
probably there were some compiled files that were not updated ? or the wrong tofu version was being used ?

Didou09, 2020-01-21:
Yes, compiled files not updated, no worries on this anymore,, see my above comment :-)