ToFuProject/tofu #329: Broken tutorial: default configuration

lasofivec, 2020-01-21:
With PR #327 the following lines were introduced

https://github.com/ToFuProject/tofu/blob/7a3aca15417e765d80f51e1585a46ebd124ca7d2/tofu/geom/utils.py#L886-L887

this breaks the [tutorial for plotting the gallery of fusion machines](https://tofuproject.github.io/tofu/auto_examples/tutorials/tuto_plot_gallery_fusion_machines.html). In particular the lines:

` config = tf.geom.utils.create_config()  # create default configuration`

We can either:
- restore a default configuration creation
- change the tutorial and plot a specific configuration

Didou09, 2020-01-21:
Good point,
I changed this to make sure that when the user does not specify a config, a new error message appears that shows all available configurations.
I think this is better since it automatically informs the user about all options he/she has.

Based on my experience that when a user meets a new function, first reflex is to to just use it without reading the docstr and see what happens.
In such a case, it should either:
* Option 1 : work out-of-the-box
   It was the case before the PR, it was automatically loading ITER. But it does not tell the user how to use the function if he/she wants to do something else
* Option 2: throw an error message that explains how to use it

We were in option 1 before the PR, we are now in option 2: the error message is explicit, shows the available config and how to load them.

Option 2 has my preference, but this is clearly a question of taste, we can go back to option 1 if you prefer, what do you think ?

lasofivec, 2020-01-21:
If the tutorial is the only place where the function was being called as such, I agree that option 2 is more tasteful.

The tutorial can call:
`config = tf.geom.utils.create_config("ITER")`

Didou09, 2020-01-21:
Good for me, can you update it ? :-)

Didou09, 2020-01-24:
Fixed in devel by PR #333

closed 2020-04-07 by Didou09