ToFuProject/tofu #331: _GG.Poly_isClockwise not working for some polygons

Didou09, 2020-01-21:
Trying to add the new geometry of ASDEX Upgrade (AUG), that a user sent to us, I ran into an issue that seems to be due to _GG.Poly_isClockwise()

I just updated branch Issue281_NewDefaultConfigs (merged with devel and pushed to github).
I also re-compiled the cython binaries.

But when trying to create the AUG config from the pre-defined txt files (containing the R, Z polygons of each structure), 3 polygons are not handled properly (the config is still created because I implemented a try/except on each structure):

```
In [1]: import tofu as tf
conf = /Home/DV226270/ToFu_All/tofu_git/tofu/tofu/imas2tofu/__init__.py:83: UserWarning: 
You do not seem to be using the latest IMAS version:
'module list' vs 'module av IMAS' suggests:
	- Current version: 3.23.1-4.0.3
	- Latest version : 3.25.0-4.4.0
  warnings.warn(msg)
/Home/DV226270/ToFu_All/tofu_git/tofu/tofu/__init__.py:95: UserWarning: 
The following subpackages are not available:
    - tofu.mag
  => see print(tofu.dsub[<subpackage>]) for details.
  warnings.warn(msg)

In [2]: conf = tf.geom.utils.create_config('AUG')
/Home/DV226270/ToFu_All/tofu_git/tofu/tofu/geom/utils.py:838: UserWarning: Could not be loaded: TFG_PFC_ExpAUG_D2dBu1.txt
  warnings.warn(msg)
/Home/DV226270/ToFu_All/tofu_git/tofu/tofu/geom/utils.py:838: UserWarning: Could not be loaded: TFG_PFC_ExpAUG_D2dBu2.txt
  warnings.warn(msg)
/Home/DV226270/ToFu_All/tofu_git/tofu/tofu/geom/utils.py:838: UserWarning: Could not be loaded: TFG_PFC_ExpAUG_SBi.txt
  warnings.warn(msg)

In [3]: conf.plot()
```

For every structure failing, there is warning. If you plot the config you'll notice it is missing some grey structures in the right hand side of the poloidal cross-section.

Could this have something to do with the bug corrected in PR #323 ?

lasofivec, 2020-01-21:
Yes it could be, quick question: why do you think it's coming from this function?
So I can save up time tracing back the issue

Didou09, 2020-01-21:
I started debugging and I stopped at this function
Sorry gotta catch the bus
Good to have you back !!! :-)

lasofivec, 2020-01-21:
> A little explanation on how the algorithm works:
> To know if a polygon is (counter) clock wise, we determine if the triangle defined by the lower left point of the polygon and its two neighbors is (counter) clock wise. 

I realized that most of the polygons that couldn't be plotted had a null volume, which would mean that the three points defining the lower left triangles are aligned, which by definition is impossible.
Just to see what would happen I said a polygon is counterclockwise even if the volume is 0.
Now more figures are plotted:

[image]

But looking a bit closer (with indices=True)

[image]

It seems to me that the point 0 is not set correctly. The algorithm finds the right lower left corner (the point 41) and measures the volume of the triangle computed by its two neighbor (normally 0 (by modulo operation) and 40) which are aligned here.

lasofivec, 2020-01-21:
PS: even by allowing to considerate a null volume to form a counterclockwise polygon, there is still one structure that is not being plotted `TFG_PFC_ExpAUG_LIM09.txt`

To help debug in the future I can: raise an exception when a null volume is found (suggesting that the polygon was ill defined). But it should be caught by the function calling the "isClockwise" function to tell the user which structure is not defined correctly. What do you think ?

Didou09, 2020-01-22:
Beware: the indices on the figure are not indices of the points, but indices of segments.
However, the conclusion remains the same, you are right, it seems the polygon is corrupted and segment 0 does not seem to be at the right place.
I'll fix it in a PR.

tofu should definitely detect such corrupted polygons and raise Exceptions explaining it to the user.
So:
* I fix the polygons (to have clean polygons)
* We finish you PR #332 , it is useful to warn the user

Didou09, 2020-01-24:
Fixed in devel by PR #332

closed 2020-04-07 by Didou09