ToFuProject/tofu #340: Anaconda deployment failing

Didou09, 2020-01-31:
Version 1.4.2 could not be deployed to anaconda on the [tagged build](https://travis-ci.org/ToFuProject/tofu/builds/644286200)

The script failed with exit status 127
I d'ont know what it corresponds to but I notice that at least 1 environment variable used in the anaconda_upload.sh script is not defined: **$PKG_DIR**

In previous versions (1.4.1) it was defined in **_install_** section as:
`export PKG_DIR=$HOME/miniconda/conda-bld/linux-64/`

I suspect this is the cause of the issue.

Other remarks:
-----------------
* Looking at the build log, it seems that the section _**before_deploy**_ is run not just once but once before each deployment.
=> As a result the  `conda build conda_recipe` is run everytime, even for pip sdist deployment, this takes a lot of time. I suggest moving this to section _**after_success**_ to make sure it is run only once.

* The version number in tofu/version.py had to be forced by hand
Indeed, the automated update in _**install**_ section (`export REV=$(python -c "import _updateversion as up; out=up.updateversion(); print(out)"`)) does ot seem to apply at deployment.
I mean that the variable REV is ok, but the version number taken at pipy deployment seems to be not $REV but instead read from tofu/version.py before it was updated by the line above. This suggest that at deployment time, travis re-clones the github repo instead of using the local tofu.

Highest priority: 
* fix anaconda_upload.sh (deployment today early afternoon if possible)
* Fix multiple `conda build conda_recipe` to save time

Lowest priority:
* tofu version update (we can work like this: i.e. by forcing version by hand for 1.4.2, but it would be nice to fix it for somehow for the next release 1.4.3)

Didou09, 2020-01-31:
I'm back from a meeting but  for not for long, I have to leave again in 5 min.
I saw you could deploy to anaconda for linux and MacOS, great ! Thanks !
Do you think some of the issues above can be solved definitely today ? And could write a short summary of what you found / understood ? I'll have  a look this week-end :-)

lasofivec, 2020-01-31:
Here's what I've managed to solve, what's left to do

- fix anaconda_upload.sh (deployment today early afternoon if possible)

Done. As you mentioned

> In previous versions (1.4.1) it was defined in install section as:
> export PKG_DIR=$HOME/miniconda/conda-bld/linux-64/

I think I didn't push a commit at some point because I remember taking care of this, but in any case now it's solved here (note that the os version is important for mac and linux deployment):

https://github.com/ToFuProject/tofu/blob/05ba14fe6ecca803dcdc14d60c88dc3143763fa3/anaconda_upload.sh#L9

- Fix multiple conda build conda_recipe to save time (should be easy)
I hadn't notice `before_deploy` was run for each provider. I didn't solve it as you mentioned it (*I suggest moving this to section after_success to make sure it is run only once.*) since this would have slowed down all the pushes that are not a release. And that was the main reason I changed it to `before_deploy`. However I found that I am not the only one with issue, and implemented the solution on this thread:

https://github.com/travis-ci/travis-ci/issues/2570

See
https://github.com/ToFuProject/tofu/blob/d37e4a41d4caca5313a74b76e4c26421f56062f1/.travis.yml#L72-L84

- The version number in tofu/version.py had to be forced by hand

Not sure what this means, you had to change the version number by hand and push it ? This sounds to me like there is an issue in our script update_version.py since now we have two versions of this functionality, I am guessing it's coming from there. If that's the case, I will take a look.

- Clarify tag vs branch condition in travis (master vs deploy-test)

Again, we are not the only ones with this issue:
https://github.com/travis-ci/travis-ci/issues/4745

I took out the `tags: true` condition for test-deploy (I will test it soon)

lasofivec, 2020-02-06:
I have been trying to understand what you meant by 

> The version number in tofu/version.py had to be forced by hand

I have been trying on a personal project as well as on the `deploy-test` branch and it seems to be working... but maybe you can explain a little bit more ?

closed 2020-04-07 by Didou09