ToFuProject/tofu #344: Deploy bin via setup.py

Didou09, 2020-02-06:
tofu typically delivers at leawt 2 bash scripts to be executable driectly from the command line:
* tofuplot : load data from imas and plot it in an interactive figure
* tofucalc : load data from imas and use it to calculate synthetic data for a diagnostic and plot it in an interactive figure

They can be used only when the sub-package imas2tofu is available (i.e.: only when imas is available), but in that case they greatly simplify calls to tofu for users not familiar with python. They do not allow as many options as calling tofu from the ipython console, but they provide the basics for an everyday use.

These 2 binaries shall be automatically deloyed by setup.py at tofu installation in order to be available straight from the command line.
We'll use the [tool ](https://python-packaging.readthedocs.io/en/latest/command-line-scripts.html) that @lasofivec  found recently (first option with script keyword arg because it allows for non-python scripts).

Didou09, 2020-02-06:
Fixed, in devel, by PR #345

lasofivec, 2020-02-06:
I changed the name to be a bit more explicit

closed 2020-04-07 by Didou09