ToFuProject/tofu #347: _checkformat_inputs_dconfig error in travis

lasofivec, 2020-02-06:
I have been playing a lot with the travis.yml but branch `deploy-test` is failing with the following error:

> ======================================================================
> FAIL: Failure: AssertionError (Arg config must have at least a StructIn subclass !)
> ----------------------------------------------------------------------
> Traceback (most recent call last):
>   File "/home/travis/miniconda/lib/python3.6/site-packages/nose/failure.py", line 39, in runTest
>     raise self.exc_val.with_traceback(self.tb)
>   File "/home/travis/miniconda/lib/python3.6/site-packages/nose/loader.py", line 417, in loadTestsFromName
>     addr.filename, addr.module)
>   File "/home/travis/miniconda/lib/python3.6/site-packages/nose/importer.py", line 47, in importFromPath
>     return self.importFromDir(dir_path, fqname)
>   File "/home/travis/miniconda/lib/python3.6/site-packages/nose/importer.py", line 94, in importFromDir
>     mod = load_module(part_fqname, fh, filename, desc)
>   File "/home/travis/miniconda/lib/python3.6/imp.py", line 235, in load_module
>     return load_source(name, filename, file)
>   File "/home/travis/miniconda/lib/python3.6/imp.py", line 172, in load_source
>     module = _load(spec)
>   File "<frozen importlib._bootstrap>", line 684, in _load
>   File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
>   File "<frozen importlib._bootstrap_external>", line 678, in exec_module
>   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
>   File "/home/travis/miniconda/lib/python3.6/site-packages/tofu-1.4.2b4-py3.6-linux-x86_64.egg/tofu/tests/tests01_geom/tests03_core.py", line 647, in <module>
>     Exp=_Exp, Diag='Test', SavePath=_here)
>   File "/home/travis/miniconda/lib/python3.6/site-packages/tofu-1.4.2b4-py3.6-linux-x86_64.egg/tofu/geom/_core.py", line 3610, in __init__
>     super(Rays, self).__init__(**kwdargs)
>   File "/home/travis/miniconda/lib/python3.6/site-packages/tofu-1.4.2b4-py3.6-linux-x86_64.egg/tofu/utils.py", line 1399, in __init__
>     self._init(**kwdargs)
>   File "/home/travis/miniconda/lib/python3.6/site-packages/tofu-1.4.2b4-py3.6-linux-x86_64.egg/tofu/geom/_core.py", line 4035, in _init
>     self.set_dconfig(calcdgeom=False, **kwdconfig)
>   File "/home/travis/miniconda/lib/python3.6/site-packages/tofu-1.4.2b4-py3.6-linux-x86_64.egg/tofu/geom/_core.py", line 4050, in set_dconfig
>     config = self._checkformat_inputs_dconfig(config)
>   File "/home/travis/miniconda/lib/python3.6/site-packages/tofu-1.4.2b4-py3.6-linux-x86_64.egg/tofu/geom/_core.py", line 3917, in _checkformat_inputs_dconfig
>     assert len(lSIn) > 0, msg
> AssertionError: Arg config must have at least a StructIn subclass !

an idea of what might cause this ?

Didou09, 2020-02-07:
That's very straneg, there is no reason this should fail, we haven't changed these tests in ages.
I notice the last tests all passed, did you change something ?

lasofivec, 2020-02-07:
yes, I install the tofu with `pip install -e .[dev]` instead of `python setup.py install` ..... ?!

Didou09, 2020-02-07:
?? ... no idea...
I'll try it locally on my working station to see

Didou09, 2020-02-25:
I Created a dedicated branch Issue347_geomtestsfailing to make the error message more complete, and pushed it.
No error was raised despite the fact that I did not modify the test, all travis builds were [successfull](https://travis-ci.org/ToFuProject/tofu/builds/654782070https://travis-ci.org/ToFuProject/tofu/builds/654782070). 
This error seems impossible to reproduce.
I am closing this issue, until it shows up again, but it does not seem to.

I'll just keep the more explicit error messages

closed 2020-02-25 by Didou09

lasofivec, 2020-03-02:
> yes, I install the tofu with `pip install -e .[dev]` instead of `python setup.py install` ..... ?!

Did you try doing this ?