ToFuProject/tofu #366: Failed deployment to anaconda.org

Didou09, 2020-03-11:
Deployment of tofu 1.4.3 was a half-success (see [build](https://travis-ci.org/github/ToFuProject/tofu/builds/661020789))  

[Pypi](https://pypi.org/project/tofu/1.4.3/#files):
------
* sdist to pypi ok
* bdist ok for mac (3.7 and 3.6)
* bdist failed for linux

[Anaconda](https://anaconda.org/ToFuProject/tofu/files):
------------
* anaconda for linux was not attempted because bdist failed for linux
* anaconda for Mac ok

Is there a way to make sure a failed deployment does not stop the other deployments (linux cases) ?

lasofivec, 2020-03-30:
For `bdist` distribution on Linux machines: as there are so many different linux distributions, with different compilers, etc. for the moment pypa doesn't distribute wheels (already compiled code) for linux machines, but somehow the conditions for the deployment were not met and it still try to deploy it as such. Solving this in a new PR

Didou09, 2020-03-30:
Fixed, in devel, by PR #371

closed 2020-04-07 by Didou09